coolsnowwolf/lede #725: 想用dnsforwarder替代pdnsd

tossp, 2018-10-22:
这个想法不知道能不能提交PR？

现在同时开启flowoffload和shadowsocksr会跑出两个pdnsd进程

pdnsd已经很久没更新，相反dnsforwarder的开发很活跃，配置也更灵活

dns缓存统一管理，个人觉得更好，但是要需要修改flowoffload和ssplus的部分代码，要不完全独立的dns cache 完全没意义。

tossp, 2018-10-23:
[image]

昨天看的Windows版的，今天编译了一个版本出来
dnsforwarder的尺寸是pdnsd 的一半。。。。
@coolsnowwolf

coolsnowwolf, 2018-10-30:
其实还是要分开合适，大小嘛。。。这几十KB 问题不大。观察 DNSFORWARDER 再成熟点吧

tossp, 2018-10-30:
@coolsnowwolf 嗯嗯，关于DNSFORWARDER 对openwrt的适配已经PR了，觉得可取的话可以合并

laincat, 2018-10-31:
说到陈旧。。。SS/SSR 客户端 真想换掉，大雕得 SSR-Pro / SSR Plus 是壳，不知道改起来麻不麻烦，或者重新造一个对应得 LUCI

客户端，如下。优点是 包含了 多种协议，SS/SSR/v2ray，等等。

https://github.com/Dreamacro/clash ，这个优点是 开始有机场兼容配置文件，一个文件实现各种分流，等等。目前也有各个平台的 各种第三方 GUI。前景可能不错。
https://github.com/nadoo/glider ，这个优点是已经稳定，并且功能比较完善，缺点是复杂的配置比较不便。目前开发有点慢。

laincat, 2018-10-31:
> 
> 
> 这个想法不知道能不能提交PR？
> 
> 现在同时开启flowoffload和shadowsocksr会跑出两个pdnsd进程
> 
> pdnsd已经很久没更新，相反dnsforwarder的开发很活跃，配置也更灵活
> 
> dns缓存统一管理，个人觉得更好，但是要需要修改flowoffload和ssplus的部分代码，要不完全独立的dns cache 完全没意义。

https://github.com/chenhw2/openwrt-dnsmasq-extra

tossp, 2018-11-01:
> 说到陈旧。。。SS/SSR 客户端 真想换掉，大雕得 SSR-Pro / SSR Plus 是壳，不知道改起来麻不麻烦，或者重新造一个对应得 LUCI
> 
> 客户端，如下。优点是 包含了 多种协议，SS/SSR/v2ray，等等。
> 
> https://github.com/Dreamacro/clash ，这个优点是 开始有机场兼容配置文件，一个文件实现各种分流，等等。目前也有各个平台的 各种第三方 GUI。前景可能不错。
> https://github.com/nadoo/glider ，这个优点是已经稳定，并且功能比较完善，缺点是复杂的配置比较不便。目前开发有点慢。

类似的项目一直没铺开，或者说一直没形成一个优秀的方案，让所有人或者说大多人都满意的

主要问题，还是在切换是不够智能，不如直接指定gfw那么流畅，负载均衡也导致了只对TCP生效，UDP完全没被考虑，目前我多机场方案用的是haporxy

laincat, 2018-11-02:
> 类似的项目一直没铺开，或者说一直没形成一个优秀的方案，让所有人或者说大多人都满意的
>
> 主要问题，还是在切换是不够智能，不如直接指定gfw那么流畅，负载均衡也导致了只对TCP生效，UDP完全没被考虑，目前我多机场方案用的是haporxy

同意，不过已经有这么多轮子了，完善还是有点希望的。

不过目前还是希望大大完成 dnsforwarder 替代计划比较快。。。感谢您的回复和科普。还有你的付出。

laincat, 2018-11-14:
"透明"是否这个更棒？
https://github.com/wongsyrone/lede-1/tree/master/package/external/transocks-wong
https://github.com/wongsyrone/transocks-wong

PR #754 (2018-11-18): 添加 Dnsforwarder 包
# 说明

添加`CONFIG_PACKAGE_luci-app-dnsforwarder`

`luci-app-dnsforwarder`依赖`dnsforwarder`

兼容`flowoffload`，详细方式查看`fixflowoffload`函数

暂时没有做多语言支持

计划下一步通过添加`GroupFile[解析策略(文件)]`参数的方式兼容ssr的gfwlist解析。

ssr的解析处理可以依赖`dnsforwarder`完成，透明翻墙依然使用`ipset`

## 界面截图

[image]

[image]

[image]

[image]

# 其他说明
closed #725
closed 2018-11-18 by coolsnowwolf