coolsnowwolf/lede #733: v2选择绕过大陆IP就没法翻墙

lucky0l, 2018-10-23:
系统日志如下，不知道帖的对不对，如果不对大雕请告诉下，我再找放入！

Tue Oct 23 21:28:41 2018 daemon.info dnsmasq[20996]: using nameserver 127.0.0.1#7453 for domain zynaima.com
Tue Oct 23 21:28:41 2018 daemon.info dnsmasq[20996]: using nameserver 127.0.0.1#7453 for domain zvereff.com
Tue Oct 23 21:28:41 2018 daemon.info dnsmasq[20996]: using nameserver 127.0.0.1#7453 for domain zuola.com
Tue Oct 23 21:28:41 2018 daemon.info dnsmasq[20996]: using nameserver 127.0.0.1#7453 for domain zuobiao.me
Tue Oct 23 21:28:41 2018 daemon.info dnsmasq[20996]: using nameserver 127.0.0.1#7453 for domain zuo.la
Tue Oct 23 21:28:41 2018 daemon.info dnsmasq[20996]: using nameserver 127.0.0.1#7453 for domain zsrhao.com
Tue Oct 23 21:28:41 2018 daemon.info dnsmasq[20996]: using nameserver 127.0.0.1#7453 for domain zspeeder.me
Tue Oct 23 21:28:41 2018 daemon.info dnsmasq[20996]: using nameserver 127.0.0.1#7453 for domain zshare.net
Tue Oct 23 21:28:41 2018 daemon.info dnsmasq[20996]: using nameserver 127.0.0.1#7453 for domain zpn.im
Tue Oct 23 21:28:41 2018 daemon.info dnsmasq[20996]: using nameserver 127.0.0.1#7453 for domain zozotown.com
Tue Oct 23 21:28:41 2018 daemon.info dnsmasq[20996]: using nameserver 127.0.0.1#7453 for domain zorrovpn.com
Tue Oct 23 21:28:41 2018 daemon.info dnsmasq[20996]: using nameserver 127.0.0.1#7453 for domain zoozle.net
Tue Oct 23 21:28:41 2018 daemon.info dnsmasq[20996]: using nameserver 127.0.0.1#7453 for domain zootool.com
Tue Oct 23 21:28:41 2018 daemon.info dnsmasq[20996]: using nameserver 127.0.0.1#7453 for domain zoogvpn.com
Tue Oct 23 21:28:41 2018 daemon.info dnsmasq[20996]: using nameserver 127.0.0.1#7453 for domain zonghexinwen.net
Tue Oct 23 21:28:41 2018 daemon.info dnsmasq[20996]: using nameserver 127.0.0.1#7453 for domain zonghexinwen.com
Tue Oct 23 21:28:41 2018 daemon.info dnsmasq[20996]: using nameserver 127.0.0.1#7453 for domain zonaeuropa.com
Tue Oct 23 21:28:41 2018 daemon.info dnsmasq[20996]: using nameserver 127.0.0.1#7453 for domain zomobo.net
Tue Oct 23 21:28:41 2018 daemon.info dnsmasq[20996]: using nameserver 127.0.0.1#7453 for domain zodgame.us
Tue Oct 23 21:28:41 2018 daemon.info dnsmasq[20996]: using nameserver 127.0.0.1#7453 for domain zmw.cn
Tue Oct 23 21:28:41 2018 daemon.info dnsmasq[20996]: using nameserver 127.0.0.1#7453 for domain zkaip.com
Tue Oct 23 21:28:41 2018 daemon.info dnsmasq[20996]: using nameserver 127.0.0.1#7453 for domain zippyshare.com
Tue Oct 23 21:28:41 2018 daemon.info dnsmasq[20996]: using nameserver 127.0.0.1#7453 for domain ziporn.com
Tue Oct 23 21:28:41 2018 daemon.info dnsmasq[20996]: using nameserver 127.0.0.1#7453 for domain ziplib.com
Tue Oct 23 21:28:41 2018 daemon.info dnsmasq[20996]: using 5882 more nameservers
Tue Oct 23 21:28:41 2018 daemon.info dnsmasq[20996]: read /etc/hosts - 4 addresses
Tue Oct 23 21:28:41 2018 daemon.info dnsmasq[20996]: read /tmp/hosts/odhcpd - 1 addresses
Tue Oct 23 21:28:41 2018 daemon.info dnsmasq[20996]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Tue Oct 23 21:28:41 2018 daemon.info dnsmasq-dhcp[20996]: read /etc/ethers - 0 addresses
Tue Oct 23 21:28:46 2018 daemon.warn dnsmasq[20996]: nameserver 192.168.1.1 refused to do a recursive query
Tue Oct 23 21:30:07 2018 daemon.info dnsmasq[20996]: exiting on receipt of SIGTERM
Tue Oct 23 21:30:07 2018 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Oct 23 21:30:07 2018 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Tue Oct 23 21:30:08 2018 daemon.info dnsmasq[21232]: started, version 2.80 cachesize 150
Tue Oct 23 21:30:08 2018 daemon.info dnsmasq[21232]: DNS service limited to local subnets
Tue Oct 23 21:30:08 2018 daemon.info dnsmasq[21232]: compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack ipset no-auth no-DNSSEC loop-detect inotify dumpfile
Tue Oct 23 21:30:08 2018 daemon.info dnsmasq-dhcp[21232]: DHCP, IP range 192.168.254.100 -- 192.168.254.249, lease time 1d
Tue Oct 23 21:30:08 2018 daemon.info dnsmasq[21232]: using local addresses only for domain test
Tue Oct 23 21:30:08 2018 daemon.info dnsmasq[21232]: using local addresses only for domain onion
Tue Oct 23 21:30:08 2018 daemon.info dnsmasq[21232]: using local addresses only for domain localhost
Tue Oct 23 21:30:08 2018 daemon.info dnsmasq[21232]: using local addresses only for domain local
Tue Oct 23 21:30:08 2018 daemon.info dnsmasq[21232]: using local addresses only for domain invalid
Tue Oct 23 21:30:08 2018 daemon.info dnsmasq[21232]: using local addresses only for domain bind
Tue Oct 23 21:30:08 2018 daemon.info dnsmasq[21232]: using local addresses only for domain lan
Tue Oct 23 21:30:08 2018 daemon.info dnsmasq[21232]: reading /tmp/resolv.conf.auto
Tue Oct 23 21:30:08 2018 daemon.info dnsmasq[21232]: using local addresses only for domain test
Tue Oct 23 21:30:08 2018 daemon.info dnsmasq[21232]: using local addresses only for domain onion
Tue Oct 23 21:30:08 2018 daemon.info dnsmasq[21232]: using local addresses only for domain localhost
Tue Oct 23 21:30:08 2018 daemon.info dnsmasq[21232]: using local addresses only for domain local
Tue Oct 23 21:30:08 2018 daemon.info dnsmasq[21232]: using local addresses only for domain invalid
Tue Oct 23 21:30:08 2018 daemon.info dnsmasq[21232]: using local addresses only for domain bind
Tue Oct 23 21:30:08 2018 daemon.info dnsmasq[21232]: using local addresses only for domain lan
Tue Oct 23 21:30:08 2018 daemon.info dnsmasq[21232]: using nameserver 192.168.1.1#53
Tue Oct 23 21:30:08 2018 daemon.info dnsmasq[21232]: read /etc/hosts - 4 addresses
Tue Oct 23 21:30:08 2018 daemon.info dnsmasq[21232]: read /tmp/hosts/odhcpd - 1 addresses
Tue Oct 23 21:30:08 2018 daemon.info dnsmasq[21232]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Tue Oct 23 21:30:08 2018 daemon.info dnsmasq-dhcp[21232]: read /etc/ethers - 0 addresses
Tue Oct 23 21:30:08 2018 cron.info crond[21281]: crond (busybox 1.29.2) started, log level 5
Tue Oct 23 21:30:08 2018 daemon.warn dnsmasq[21232]: nameserver 192.168.1.1 refused to do a recursive query
Tue Oct 23 21:30:10 2018 daemon.info pdnsd[21738]: pdnsd-1.2.9b-par starting.
Tue Oct 23 21:30:10 2018 daemon.warn pdnsd[21738]: Cache file /var/pdnsd/pdnsd.cache ignored because of incompatible version identifier
Tue Oct 23 21:30:10 2018 daemon.info dnsmasq[21232]: exiting on receipt of SIGTERM
Tue Oct 23 21:30:10 2018 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Oct 23 21:30:11 2018 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Tue Oct 23 21:30:11 2018 daemon.info dnsmasq[21820]: started, version 2.80 cachesize 150
Tue Oct 23 21:30:11 2018 daemon.info dnsmasq[21820]: DNS service limited to local subnets
Tue Oct 23 21:30:11 2018 daemon.info dnsmasq[21820]: compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack ipset no-auth no-DNSSEC loop-detect inotify dumpfile
Tue Oct 23 21:30:11 2018 daemon.info dnsmasq-dhcp[21820]: DHCP, IP range 192.168.254.100 -- 192.168.254.249, lease time 1d
[... 618 lines cut ...]
Tue Oct 23 21:36:50 2018 daemon.info dnsmasq[26040]: using nameserver 127.0.0.1#7453 for domain zsrhao.com
Tue Oct 23 21:36:50 2018 daemon.info dnsmasq[26040]: using nameserver 127.0.0.1#7453 for domain zspeeder.me
Tue Oct 23 21:36:50 2018 daemon.info dnsmasq[26040]: using nameserver 127.0.0.1#7453 for domain zshare.net
Tue Oct 23 21:36:50 2018 daemon.info dnsmasq[26040]: using nameserver 127.0.0.1#7453 for domain zpn.im
Tue Oct 23 21:36:50 2018 daemon.info dnsmasq[26040]: using nameserver 127.0.0.1#7453 for domain zozotown.com
Tue Oct 23 21:36:50 2018 daemon.info dnsmasq[26040]: using nameserver 127.0.0.1#7453 for domain zorrovpn.com
Tue Oct 23 21:36:50 2018 daemon.info dnsmasq[26040]: using nameserver 127.0.0.1#7453 for domain zoozle.net
Tue Oct 23 21:36:50 2018 daemon.info dnsmasq[26040]: using nameserver 127.0.0.1#7453 for domain zootool.com
Tue Oct 23 21:36:50 2018 daemon.info dnsmasq[26040]: using nameserver 127.0.0.1#7453 for domain zoogvpn.com
Tue Oct 23 21:36:50 2018 daemon.info dnsmasq[26040]: using nameserver 127.0.0.1#7453 for domain zonghexinwen.net
Tue Oct 23 21:36:50 2018 daemon.info dnsmasq[26040]: using nameserver 127.0.0.1#7453 for domain zonghexinwen.com
Tue Oct 23 21:36:50 2018 daemon.info dnsmasq[26040]: using nameserver 127.0.0.1#7453 for domain zonaeuropa.com
Tue Oct 23 21:36:50 2018 daemon.info dnsmasq[26040]: using nameserver 127.0.0.1#7453 for domain zomobo.net
[... cut ...]
Tue Oct 23 21:36:50 2018 daemon.info dnsmasq[26040]: using 5883 more nameservers
Tue Oct 23 21:36:50 2018 daemon.info dnsmasq[26040]: read /etc/hosts - 4 addresses
Tue Oct 23 21:36:50 2018 daemon.info dnsmasq[26040]: read /tmp/hosts/odhcpd - 1 addresses
Tue Oct 23 21:36:50 2018 daemon.info dnsmasq[26040]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Tue Oct 23 21:36:50 2018 daemon.info dnsmasq-dhcp[26040]: read /etc/ethers - 0 addresses
Tue Oct 23 21:38:35 2018 daemon.warn dnsmasq[26040]: nameserver 192.168.1.1 refused to do a recursive query

lucky0l, 2018-10-23:
内核日志

[    0.000000] Linux version 4.14.78 (lean@LAPTOP-Lucky) (gcc version 7.3.0 (OpenWrt GCC 7.3.0 r888-2deb00db)) #0 SMP Mon Oct 22 17:18:28 2018
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is UBNT-ERX
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 10000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] On node 0 totalpages: 65536
[    0.000000] free_area_init_node: node 0, pgdat 8051da80, node_mem_map 81003000
[    0.000000]   Normal zone: 512 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000] random: get_random_bytes called from 0x80521740 with crng_init=0
[    0.000000] percpu: Embedded 14 pages/cpu @8120f000 s26000 r8192 d23152 u57344
[    0.000000] pcpu-alloc: s26000 r8192 d23152 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 65024
[    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Writing ErrCtl register=00039200
[    0.000000] Readback ErrCtl register=00039200
[    0.000000] Memory: 253836K/262144K available (4467K kernel code, 228K rwdata, 544K rodata, 252K init, 244K bss, 8308K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 256
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.000010] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
[    0.015490] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.087814] pid_max: default: 32768 minimum: 301
[    0.097129] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.110148] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.125842] Hierarchical SRCU implementation.
[    0.135294] smp: Bringing up secondary CPUs ...
[    0.144826] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.144836] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.144847] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.144980] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.204674] Synchronize counters for CPU 1: done.
[    0.268593] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.268601] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.268609] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.268678] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.325485] Synchronize counters for CPU 2: done.
[    0.385639] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.385646] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.385654] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.385735] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.445053] Synchronize counters for CPU 3: done.
[    0.504650] smp: Brought up 1 node, 4 CPUs
[    0.516060] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.535526] futex hash table entries: 1024 (order: 3, 32768 bytes)
[    0.547931] pinctrl core: initialized pinctrl subsystem
[    0.559297] NET: Registered protocol family 16
[... 108 lines cut ...]
[   13.128990] Netfilter messages via NETLINK v0.30.
[   13.140219] ip_set: protocol 6
[   13.178393] u32 classifier
[   13.183808]     input device check on
[   13.191331]     Actions configured
[   13.200490] Mirror/redirect action on
[   13.217896] nf_conntrack version 0.5.0 (4096 buckets, 16384 max)
[   13.238413] ip_tables: (C) 2000-2006 Netfilter Core Team
[   13.288372] ctnetlink v0.93: registering with nfnetlink.
[   13.396903] xt_time: kernel timezone is -0000
[   13.420240] PPP generic driver version 2.4.2
[   13.430921] NET: Registered protocol family 24
[   13.443886] kmodloader: done loading kernel modules from /etc/modules.d/*
[   14.733116] urandom_read: 5 callbacks suppressed
[   14.733127] random: jshn: uninitialized urandom read (4 bytes read)
[   20.852699] random: crng init done
[   25.487724] mtk_soc_eth 1e100000.ethernet: PPE started
[   25.507552] br-lan: port 1(eth0.1) entered blocking state
[   25.518341] br-lan: port 1(eth0.1) entered disabled state
[   25.529921] device eth0.1 entered promiscuous mode
[   25.539529] device eth0 entered promiscuous mode
[   25.556587] br-lan: port 1(eth0.1) entered blocking state
[   25.567446] br-lan: port 1(eth0.1) entered forwarding state
[   25.579046] IPv6: ADDRCONF(NETDEV_UP): br-lan: link is not ready
[   26.529267] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready

fy1128, 2018-10-23:
嗯，一直都是。

只可以 gfwlist

lan542662, 2018-10-29:
我刚好和你相反啊...

mylukin, 2018-11-03:
我也遇到一样的问题，绕过大陆IP就无法翻墙。

lucky0l, 2018-11-19:
@mylukin 谢谢，晚些试一下，这个自己修改自己本地的试试

lucky0l, 2018-11-20:
@mylukin 我遇到了个问题，我把这一行前面加#注释掉，之后编译是成功的，但是V2反而无法运行了，你有这个问题么？

mylukin, 2018-11-20:
@lucky0l 没有遇到这个问题，不过我发现有一些防火墙的奇怪现象，所以我把v2ray的443端口改成了444端口，就没问题了。

lan542662, 2018-11-20:
我去。那这个问题不是和sspro刚好相反吗。sspro是不能gfw莫名其妙有人就上不了。只能绕过。这个却只能gfw...

lucky0l, 2018-11-21:
@mylukin 但是我的V2走的是TLS要用443端口，所以，估计没法改到444吧

lucky0l, 2018-11-21:
@lan542662 SSPRO我没用，所以不知道他的情况了……

mylukin, 2018-11-21:
> @mylukin 但是我的V2走的是TLS要用443端口，所以，估计没法改到444吧

TLS可以用任何端口，不限制443

closed 2018-11-27 by coolsnowwolf