coolsnowwolf/lede #738: frp无法登录，一直提示tcp_mux不一致，但在路由器下的主机测试frps服务端是正常的，没有问题。

ChuckPeng, 2018-10-24:
frp服务端frps log
2018/10/22 11:22:33 [I] [service.go:128] frps tcp listen on 0.0.0.0:xxx
2018/10/22 11:22:33 [I] [service.go:137] frps kcp listen on udp 0.0.0.0:xxx
2018/10/22 11:22:33 [I] [service.go:161] http service listen on 0.0.0.0:80
2018/10/22 11:22:33 [I] [service.go:182] https service listen on 0.0.0.0:443
2018/10/22 11:22:33 [I] [service.go:205] Dashboard listen on 0.0.0.0:xxx
2018/10/22 11:22:33 [I] [root.go:190] Start frps success
2018/10/22 11:43:00 [W] [newhttp.go:202] http: proxy error: no such domain
2018/10/22 11:44:11 [W] [newhttp.go:202] http: proxy error: no such domain
2018/10/22 11:45:13 [I] [service.go:305] client login info: ip [x.x.x.x:6                                                   4313] version [0.21.0] hostname [] os [windows] arch [amd64]
2018/10/22 11:45:13 [D] [control.go:135] [3d0e7d18fe49fe40] new work connection                                                    registered
2018/10/22 11:45:13 [D] [control.go:135] [3d0e7d18fe49fe40] new work connection                                                    registered

openwrt 8.1.8 log
2018/10/22 11:47:55: frp服务启动失败，请检查服务端 “TCP多路复用(tcp_mux)”设置，确保与客户端完全一致！

在openwrt下用windows的frpc连接frps服务端，从log看提示成功，openwrt8.1.8只提示TCP多路复用与客户端不一致，frps无任何log错误信息。无论frps或frpc是否勾选tcp复用选项都提示这个错误。

tossp, 2018-10-24:
直接查配置文件

ChuckPeng, 2018-10-24:
>           直接查配置文件

openwrt下的frp配置文件？

tossp, 2018-10-24:
是的，看看luci生成的配置文件是什么样子的

ChuckPeng, 2018-10-25:
>           是的，看看luci生成的配置文件是什么样子的

生成的配置在什么地方？搜索了下没搜到

Lancenas, 2018-10-25:
时间同步吗？

mjyhj, 2018-11-14:
tcp_mux
需要客户端和服务端一致

closed 2018-11-27 by coolsnowwolf

moxuanyuan, 2020-12-08:
同样问题，解决不了。。

iGocsen, 2024-04-14:
貌似是时间同步的问题，当日志时间为北京时间-8hours时时可以连接上的，当我时间调为北京时间时就又连不上了，日志如下：
2024/04/14 10:08:15: frp服务启动失败，请检查服务端 “TCP多路复用(tcp_mux)”设置，确保与客户端完全一致！
2024/04/14 02:08:18 [I] [service.go:304] [1cxxxeb] login to server success, get run id [1cxxxeb], server udp port [0]
2024/04/14 02:08:18 [I] [proxy_manager.go:144] [1cxxxeb] proxy added: [OpenWrt.Openwrt-Web]
2024/04/14 02:08:18 [W] [control.go:178] [1cxxxeb] [OpenWrt.Openwrt-Web] start error: router config conflict
2024/04/14 02:08:48 [I] [control.go:180] [1cxxxeb] [OpenWrt.Openwrt-Web] start proxy success
2024/04/14 02:17:51 [E] [control.go:158] [1cxxxeb] work connection closed before response StartWorkConn message: EOF
2024/04/14 02:17:51 [I] [control.go:282] [1cxxxeb] control writer is closing
2024/04/14 02:17:51 [I] [service.go:177] [1cxxxeb] try to reconnect to server...
2024/04/14 02:17:51 [I] [visitor_manager.go:60] [1cxxxeb] gracefully shutdown visitor manager
2024/04/14 02:17:51 [W] [service.go:180] [1cxxxeb] reconnect to server error: dial tcp IP:7000: connect: connection refused
2024/04/14 02:17:52 [I] [service.go:177] [1cxxxeb] try to reconnect to server...
2024/04/14 02:17:53 [I] [service.go:304] [1cxxxeb] login to server success, get run id [1cxxxeb], server udp port [0]
2024/04/14 02:17:53 [I] [proxy_manager.go:144] [1cxxxeb] proxy added: [OpenWrt.Openwrt-Web]
2024/04/14 02:17:53 [I] [control.go:180] [1cxxxeb] [OpenWrt.Openwrt-Web] start proxy success
2024/04/14 10:34:30: frp服务启动失败，请检查服务端 “TCP多路复用(tcp_mux)”设置，确保与客户端完全一致！