coolsnowwolf/lede #739: 安装curl之后，使用时报错

rengame, 2018-10-24:
opkg update 
opkg install curl

curl ip.cn

报如下错误：
Error loading shared library libmbedtls.so.10: No such file or directory (needed by /usr/bin/curl)
Error loading shared library libmbedcrypto.so.1: No such file or directory (needed by /usr/bin/curl)
Error loading shared library libmbedtls.so.10: No such file or directory (needed by /usr/lib/libcurl.so.4)
Error loading shared library libmbedcrypto.so.1: No such file or directory (needed by /usr/lib/libcurl.so.4)
Error relocating /usr/lib/libcurl.so.4: mbedtls_ssl_session_free: symbol not found
Error relocating /usr/lib/libcurl.so.4: mbedtls_ssl_init: symbol not found
Error relocating /usr/lib/libcurl.so.4: mbedtls_ssl_conf_min_version: symbol not found
Error relocating /usr/lib/libcurl.so.4: mbedtls_ssl_config_defaults: symbol not found
Error relocating /usr/lib/libcurl.so.4: mbedtls_ssl_config_init: symbol not found
Error relocating /usr/lib/libcurl.so.4: mbedtls_ssl_get_session: symbol not found
Error relocating /usr/lib/libcurl.so.4: mbedtls_ssl_free: symbol not found
Error relocating /usr/lib/libcurl.so.4: mbedtls_ssl_conf_max_version: symbol not found
Error relocating /usr/lib/libcurl.so.4: mbedtls_ssl_config_free: symbol not found
Error relocating /usr/lib/libcurl.so.4: mbedtls_ssl_get_bytes_avail: symbol not found
Error relocating /usr/lib/libcurl.so.4: mbedtls_ssl_get_peer_cert: symbol not found
Error relocating /usr/lib/libcurl.so.4: mbedtls_ssl_setup: symbol not found
Error relocating /usr/lib/libcurl.so.4: mbedtls_ssl_handshake: symbol not found
Error relocating /usr/lib/libcurl.so.4: mbedtls_ssl_conf_authmode: symbol not found
Error relocating /usr/lib/libcurl.so.4: mbedtls_ssl_set_bio: symbol not found
Error relocating /usr/lib/libcurl.so.4: mbedtls_ssl_get_verify_result: symbol not found
Error relocating /usr/lib/libcurl.so.4: mbedtls_ssl_read: symbol not found
Error relocating /usr/lib/libcurl.so.4: mbedtls_ssl_set_session: symbol not found
Error relocating /usr/lib/libcurl.so.4: mbedtls_ssl_conf_ciphersuites: symbol not found
Error relocating /usr/lib/libcurl.so.4: mbedtls_ssl_conf_rng: symbol not found
Error relocating /usr/lib/libcurl.so.4: mbedtls_ssl_list_ciphersuites: symbol not found
Error relocating /usr/lib/libcurl.so.4: mbedtls_ssl_set_hostname: symbol not found
Error relocating /usr/lib/libcurl.so.4: mbedtls_ssl_session_init: symbol not found
Error relocating /usr/lib/libcurl.so.4: mbedtls_ssl_write: symbol not found
Error relocating /usr/lib/libcurl.so.4: mbedtls_net_recv: symbol not found
Error relocating /usr/lib/libcurl.so.4: mbedtls_ssl_conf_own_cert: symbol not found
Error relocating /usr/lib/libcurl.so.4: mbedtls_ssl_conf_ca_chain: symbol not found
Error relocating /usr/lib/libcurl.so.4: mbedtls_ssl_conf_cert_profile: symbol not found
Error relocating /usr/lib/libcurl.so.4: mbedtls_net_send: symbol not found
root@OpenWrt:~#

hcym, 2018-10-24:
滑稽，一直都编译这个包，从来没用过，看来以后还是不要的好

coolsnowwolf, 2018-10-24:
不要安装，mbedtls的版本和 trunk 的版本不同了，编译时加入

rengame, 2018-10-25:
> 不要安装，mbedtls的版本和 trunk 的版本不同了，编译时加入

好的，谢谢。重新编译了。

closed 2018-10-25 by rengame

PureWhiteWu, 2018-10-29:
@coolsnowwolf hi，这个问题有办法解决么？我刷了今天编译的系统，也遇到了这个问题。

searene, 2019-04-14:
我也想知道怎么解决，今天也遇到了这个问题

searene, 2019-04-14:
解决了，执行命令opkg install libmbedtls就行了

hyue418, 2024-11-25:
> 解决了，执行命令opkg install libmbedtls就行了

点赞啊！