coolsnowwolf/lede #747: ssr gfw阻塞,现在彻底上不了.有日志

lan542662, 2018-10-28:
Sun Oct 28 16:07:38 2018 kern.info kernel: [   43.889047] IPv6: ADDRCONF(NETDEV_UP): eth0.2: link is not ready
Sun Oct 28 16:07:38 2018 daemon.notice procd: /etc/rc.d/S60flowoffload: net.ipv4.tcp_congestion_control = cubic
Sun Oct 28 16:07:38 2018 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1177]: exiting on receipt of SIGTERM
Sun Oct 28 16:07:38 2018 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Oct 28 16:07:38 2018 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Oct 28 16:07:38 2018 user.notice mac80211: Failed command: iw phy phy1 set antenna all all
Sun Oct 28 16:07:38 2018 daemon.notice netifd: radio1 (1348): command failed: Not supported (-95)
Sun Oct 28 16:07:38 2018 user.notice mac80211: Failed command: iw phy phy1 set distance 0
Sun Oct 28 16:07:38 2018 user.notice mac80211: Failed command: iw phy phy0 set antenna all all
Sun Oct 28 16:07:38 2018 daemon.notice netifd: radio0 (1347): command failed: Not supported (-95)
Sun Oct 28 16:07:38 2018 user.notice mac80211: Failed command: iw phy phy0 set distance 0
Sun Oct 28 16:07:38 2018 daemon.notice procd: /etc/rc.d/S60flowoffload: DNSMASQ change
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: started, version 2.80 cachesize 150
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: DNS service limited to local subnets
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack ipset no-auth no-DNSSEC loop-detect inotify dumpfile
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq-dhcp[1693]: DHCP, IP range 192.168.1.100 -- 192.168.1.249, lease time 1d
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using local addresses only for domain test
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using local addresses only for domain onion
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using local addresses only for domain localhost
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using local addresses only for domain local
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using local addresses only for domain invalid
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using local addresses only for domain bind
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using nameserver 127.0.0.1#7453 for domain zlib.net
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using nameserver 127.0.0.1#7453 for domain zkaip.com
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using nameserver 127.0.0.1#7453 for domain zippyshare.com
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using nameserver 127.0.0.1#7453 for domain zh.wiktionary.org
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using nameserver 127.0.0.1#7453 for domain zh.wikivoyage.org
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using nameserver 127.0.0.1#7453 for domain zh.wikisource.org
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using nameserver 127.0.0.1#7453 for domain zh.wikipedia.org
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using nameserver 127.0.0.1#7453 for domain zh.wikinews.org
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using nameserver 127.0.0.1#7453 for domain zh.uncyclopedia.wikia.com
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using nameserver 127.0.0.1#7453 for domain zh.pokerstrategy.com
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using nameserver 127.0.0.1#7453 for domain zh.m.wikipedia.org
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using nameserver 127.0.0.1#7453 for domain zh.ecdm.wikia.com
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using nameserver 127.0.0.1#7453 for domain zh-yue.wikipedia.org
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using nameserver 127.0.0.1#7453 for domain zapto.org
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using nameserver 127.0.0.1#7453 for domain zaobao.com.sg
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using nameserver 127.0.0.1#7453 for domain zaobao.com
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using nameserver 127.0.0.1#7453 for domain zannel.com
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using nameserver 127.0.0.1#7453 for domain zalmos.com
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using nameserver 127.0.0.1#7453 for domain zacebook.com
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using nameserver 127.0.0.1#7453 for domain za.mirror.archlinux-br.org
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using nameserver 127.0.0.1#7453 for domain yzzk.com
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using nameserver 127.0.0.1#7453 for domain yyii.org
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using nameserver 127.0.0.1#7453 for domain yx51.net
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using nameserver 127.0.0.1#7453 for domain ytimg.com
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using nameserver 127.0.0.1#7453 for domain ytht.net
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using nameserver 127.0.0.1#7453 for domain yt.be
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using nameserver 127.0.0.1#7453 for domain youxu.info
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using nameserver 127.0.0.1#7453 for domain youwin.com
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using nameserver 127.0.0.1#7453 for domain youversion.com
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using nameserver 127.0.0.1#7453 for domain youtube.com
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: using 3894 more nameservers
Sun Oct 28 16:07:38 2018 daemon.warn dnsmasq[1693]: no servers found in /tmp/resolv.conf.auto, will retry
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: read /etc/hosts - 4 addresses
Sun Oct 28 16:07:38 2018 daemon.info dnsmasq[1693]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Sun Oct 28 16:07:39 2018 daemon.info dnsmasq-dhcp[1693]: read /etc/ethers - 0 addresses
Sun Oct 28 16:07:39 2018 daemon.err hostapd: Configuration file: /var/run/hostapd-phy0.conf
Sun Oct 28 16:07:39 2018 daemon.err hostapd: Configuration file: /var/run/hostapd-phy1.conf
Sun Oct 28 16:07:39 2018 kern.info kernel: [   44.873231] ipq806x-gmac-dwmac 37400000.ethernet eth1: Link is Up - 1Gbps/Full - flow control off
Sun Oct 28 16:07:39 2018 kern.info kernel: [   44.953228] ipq806x-gmac-dwmac 37200000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
Sun Oct 28 16:07:39 2018 daemon.info dnsmasq[1693]: read /etc/hosts - 4 addresses
Sun Oct 28 16:07:39 2018 daemon.info dnsmasq[1693]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Sun Oct 28 16:07:39 2018 daemon.info dnsmasq-dhcp[1693]: read /etc/ethers - 0 addresses
Sun Oct 28 16:07:44 2018 kern.err /usr/sbin/ssr-redir[678]: failed to resolve server name, wait 4 seconds
Sun Oct 28 16:07:45 2018 kern.info kernel: [   50.779496] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
Sun Oct 28 16:07:45 2018 kern.info kernel: [   50.780145] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Sun Oct 28 16:07:45 2018 daemon.notice netifd: Network device 'eth0' link is up
Sun Oct 28 16:07:45 2018 daemon.notice netifd: VLAN 'eth0.2' link is up
[... 638 lines cut ...]
Sun Oct 28 16:10:54 2018 daemon.info dnsmasq[8807]: using nameserver 127.0.0.1#7453 for domain yx51.net
Sun Oct 28 16:10:54 2018 daemon.info dnsmasq[8807]: using nameserver 127.0.0.1#7453 for domain ytimg.com
Sun Oct 28 16:10:54 2018 daemon.info dnsmasq[8807]: using nameserver 127.0.0.1#7453 for domain ytht.net
Sun Oct 28 16:10:54 2018 daemon.info dnsmasq[8807]: using nameserver 127.0.0.1#7453 for domain yt.be
Sun Oct 28 16:10:54 2018 daemon.info dnsmasq[8807]: using nameserver 127.0.0.1#7453 for domain youxu.info
Sun Oct 28 16:10:54 2018 daemon.info dnsmasq[8807]: using nameserver 127.0.0.1#7453 for domain youwin.com
Sun Oct 28 16:10:54 2018 daemon.info dnsmasq[8807]: using nameserver 127.0.0.1#7453 for domain youversion.com
Sun Oct 28 16:10:54 2018 daemon.info dnsmasq[8807]: using nameserver 127.0.0.1#7453 for domain youtube.com
Sun Oct 28 16:10:54 2018 daemon.info dnsmasq[8807]: using 3896 more nameservers
Sun Oct 28 16:10:54 2018 daemon.info dnsmasq[8807]: read /etc/hosts - 4 addresses
Sun Oct 28 16:10:54 2018 daemon.info dnsmasq[8807]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Sun Oct 28 16:10:54 2018 daemon.info dnsmasq-dhcp[8807]: read /etc/ethers - 0 addresses
[... cut ...]
Sun Oct 28 16:11:02 2018 kern.warn kernel: [  216.679738] Set china-banned is full, maxelem 65536 reached
Sun Oct 28 16:11:02 2018 kern.warn kernel: [  216.686519] Set china-banned is full, maxelem 65536 reached
Sun Oct 28 16:11:02 2018 kern.warn kernel: [  216.691229] Set china-banned is full, maxelem 65536 reached
Sun Oct 28 16:11:02 2018 kern.warn kernel: [  216.697143] Set china-banned is full, maxelem 65536 reached
Sun Oct 28 16:11:15 2018 daemon.err uhttpd[1331]: luci: accepted login on / for root from 192.168.1.117

hcym, 2018-10-28:
完全不懂阻塞，完全没有问题

lan542662, 2018-10-29:
> 
> 
> 完全不懂阻塞，完全没有问题

现在换成ssrplus又没问题了.
之前是ssrplus老是上不了国内网络,很慢,ssrpro没问题,现在又反过来了.ssrplus没问题了.ssrpro走不了gfw了.emmm

closed 2018-11-27 by coolsnowwolf