coolsnowwolf/lede #752: x86-64 编译选择生成EFI GRUB images时报grub-mkimage-efi: error: relocation 0x4 is not implemented yet

colinhd8, 2018-10-30:
OS: Ubuntu 18.04.1
openwrt: 从仓库拉取当前最新版
kernel: 4.14.78
编译生成x86-64固件时，如果勾选Build EFI GRUB images (Linux x86 or x86_64 host only)，到最后阶段会报
grub-mkimage-efi: error: relocation 0x4 is not implemented yet，
然后我单独执行以下命令:
`./staging_dir/host/bin/grub-mkimage-efi -d /home/colin/softwares/lede/staging_dir/host/lib/grub/x86_64-efi -o /home/colin/softwares/lede/build_dir/target-x86_64_musl/linux-x86_64/grub2.efi/efi/boot/bootx64.efi -O x86_64-efi -p /efi/boot -c /home/colin/softwares/lede/build_dir/target-x86_64_musl/linux-x86_64/grub2.efi/efi/boot/grub.cfg --verbose boot chain configfile ext2 linux ls part_msdos reboot serial part_gpt part_msdos search fat exfat ext2 efi_gop efi_uga gfxterm`
会出现这样的错误：
`./staging_dir/host/bin/grub-mkimage-efi: info: the size of config file is 0x54.
./staging_dir/host/bin/grub-mkimage-efi: info: the total module size is 0x9d3c4.
./staging_dir/host/bin/grub-mkimage-efi: info: reading /home/colin/softwares/lede/staging_dir/host/lib/grub/x86_64-efi/kernel.img.
./staging_dir/host/bin/grub-mkimage-efi: error: invalid ELF header.
colin@ubuntu-server-3:~/softwares/lede$ ./staging_dir/host/bin/grub-mkimage-efi -d /home/colin/softwares/lede/staging_dir/host/lib/grub/x86_64-efi -o /home/colin/softwares/lede/build_dir/target-x86_64_musl/linux-x86_64/grub2.efi/efi/boot/bootx64.efi -O x86_64-efi -p /efi/boot -c /home/colin/softwares/lede/build_dir/target-x86_64_musl/linux-x86_64/grub2.efi/efi/boot/grub.cfg --verbose boot chain configfile ext2 linux ls part_msdos reboot serial part_gpt part_msdos search fat exfat ext2 efi_gop efi_uga gfxterm
./staging_dir/host/bin/grub-mkimage-efi: info: the size of config file is 0x58.
./staging_dir/host/bin/grub-mkimage-efi: info: the total module size is 0x9d3d8.
./staging_dir/host/bin/grub-mkimage-efi: info: reading /home/colin/softwares/lede/staging_dir/host/lib/grub/x86_64-efi/kernel.img.
./staging_dir/host/bin/grub-mkimage-efi: info: locating the section .text at 0x0.
./staging_dir/host/bin/grub-mkimage-efi: info: locating the section .rodata at 0xca00.
./staging_dir/host/bin/grub-mkimage-efi: info: locating the section .rodata.str1.8 at 0xcfd8.
./staging_dir/host/bin/grub-mkimage-efi: info: locating the section .rodata.str1.1 at 0xd6d7.
./staging_dir/host/bin/grub-mkimage-efi: info: locating the section .data at 0xea60.
./staging_dir/host/bin/grub-mkimage-efi: info: locating the section .module_license at 0xfcc0.
./staging_dir/host/bin/grub-mkimage-efi: info: locating the section .bss at 0xfce0.
./staging_dir/host/bin/grub-mkimage-efi: info: locating  at 0x400 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating  at 0xce00 (0xce00).
./staging_dir/host/bin/grub-mkimage-efi: info: locating  at 0xd3d8 (0xd3d8).
./staging_dir/host/bin/grub-mkimage-efi: info: locating  at 0xdad7 (0xdad7).
./staging_dir/host/bin/grub-mkimage-efi: info: locating  at 0xee60 (0xee60).
./staging_dir/host/bin/grub-mkimage-efi: info: locating  at 0x100e0 (0x100e0).
./staging_dir/host/bin/grub-mkimage-efi: info: locating  at 0x100c0 (0x100c0).
./staging_dir/host/bin/grub-mkimage-efi: info: locating  at 0x400 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_strlen at 0xb7c0 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_net_poll_cards_idle at 0x11e60 (0x100e0).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_efi_finish_boot_services at 0x3f40 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_disk_get_size at 0x74c0 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_file_progress_hook at 0x1bfa0 (0x100e0).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_efidisk_fini at 0xb40 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_puts_ at 0xa140 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_millisleep at 0x5c80 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_fs_autoload_hook at 0x11c28 (0x100e0).
./staging_dir/host/bin/grub-mkimage-efi: info: locating efi_wrap_10 at 0x654 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_byte_checksum at 0x5030 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_fs_blocklist at 0xf000 (0xee60).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_errmsg at 0x1be80 (0x100e0).
./staging_dir/host/bin/grub-mkimage-efi: info: locating efi_wrap_5 at 0x5eb (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_strncmp at 0xa220 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_strtoull at 0xa3d0 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_dma_get_virt at 0x770 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_efi_system_table at 0x11e70 (0x100e0).
./staging_dir/host/bin/grub-mkimage-efi: info: locating memmove at 0x5470 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating efi_wrap_4 at 0x5d7 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_disk_open at 0x6d70 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_efi_locate_protocol at 0x2280 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_env_update_get_sorted at 0x89a0 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_strcpy at 0xa120 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_partition_probe at 0xc010 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_strrchr at 0xa2c0 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_partition_get_name at 0xc280 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_dl_load at 0x84f0 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_pmtimer_wait_count_tsc at 0x4a0 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_efi_stall at 0x24e0 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_efi_get_filename at 0x2c40 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_env_export at 0x8ac0 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_xvasprintf at 0xb8a0 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_error_push at 0x8ba0 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_rescue_run at 0xc8f0 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_xasprintf at 0xb980 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_realloc at 0x5b70 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_exit at 0x2530 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_tsc_calibrate_from_pmtimer at 0x530 (0x400).
[... 138 lines cut ...]
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_device_iterate at 0x6670 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_getkey_noblock at 0xcae0 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_memalign_dma32 at 0x6f0 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_list_push at 0x9790 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_efi_set_text_mode at 0x2450 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_err_printf at 0xb590 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_disk_cache_invalidate_all at 0x6bf0 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_machine_acpi_get_rsdpv2 at 0x5210 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_env_set at 0x8730 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_dl_load_core at 0x83c0 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_gettext at 0xf050 (0xee60).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_memcmp at 0xa180 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_env_get at 0x88c0 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating efi_wrap_2 at 0x5b8 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_strtoul at 0xa580 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_dma_get_phys at 0x780 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_mm_init_region at 0x56c0 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_disk_cache_table at 0x11ee0 (0x100e0).
./staging_dir/host/bin/grub-mkimage-efi: info: locating grub_memalign at 0x5880 (0x400).
./staging_dir/host/bin/grub-mkimage-efi: info: dealing with the relocation section .rela.text for .text.
./staging_dir/host/bin/grub-mkimage-efi: info: relocating an R_X86_64_PC32 entry to 0x11a71 at the offset 0x3.
./staging_dir/host/bin/grub-mkimage-efi: info: relocating an R_X86_64_PC32 entry to 0x11a62 at the offset 0xa.
./staging_dir/host/bin/grub-mkimage-efi: error: relocation 0x4 is not implemented yet.`

感觉像是这个命令对最新的kernel支持不好。

Fyred, 2018-10-30:
1. 首先装好 Ubuntu 64bit，推荐  Ubuntu  14 LTS x64

coolsnowwolf, 2018-10-30:
这个是 18.04 的问题。。。