coolsnowwolf/lede #756: softethervpn5能帮忙更新到v5.01.9662吗

liuluo1979, 2018-10-31:
PKG_SOURCE_URL:=https://github.com/SoftEtherVPN/SoftEtherVPN.git
PKG_SOURCE_DATE:=2018-10-03
PKG_SOURCE_VERSION:=2271bf5fba410f815bf6834aae65d4b513cee1b2

直接改成最新的v5.01.9662编译失败
PKG_SOURCE_DATE:=2018-10-31
PKG_SOURCE_VERSION:=080edcdb8155ebaf110dfab1bd747a7fd0728203

tossp, 2018-11-05:
搞了一天，能力有限，搞不动了，一直卡在一个共享库错误`libncurses.a`
```
/builds/zh/lede-k3/staging_dir/toolchain-arm_cortex-a9_gcc-7.3.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-gcc -fPIC -Os -pipe -mcpu=cortex-a9 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=soft -iremap/builds/zh/lede-k3/build_dir/target-arm_cortex-a9_musl_eabi/libubox-2018-07-25-c83a84af:libubox-2018-07-25-c83a84af -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/builds/zh/lede-k3/staging_dir/target-arm_cortex-a9_musl_eabi/usr/include -DNDEBUG -L/builds/zh/lede-k3/staging_dir/target-arm_cortex-a9_musl_eabi/usr/lib -L/builds/zh/lede-k3/staging_dir/target-arm_cortex-a9_musl_eabi/lib -L/builds/zh/lede-k3/staging_dir/toolchain-arm_cortex-a9_gcc-7.3.0_musl_eabi/usr/lib -L/builds/zh/lede-k3/staging_dir/toolchain-arm_cortex-a9_gcc-7.3.0_musl_eabi/lib -znow -zrelro -Wl,-Bsymbolic-functions -shared -Wl,-soname,libubox.so -o libubox.so CMakeFiles/ubox.dir/avl.c.o CMakeFiles/ubox.dir/avl-cmp.c.o CMakeFiles/ubox.dir/blob.c.o CMakeFiles/ubox.dir/blobmsg.c.o CMakeFiles/ubox.dir/uloop.c.o CMakeFiles/ubox.dir/usock.c.o CMakeFiles/ubox.dir/ustream.c.o CMakeFiles/ubox.dir/ustream-fd.c.o CMakeFiles/ubox.dir/vlist.c.o CMakeFiles/ubox.dir/utils.c.o CMakeFiles/ubox.dir/safe_list.c.o CMakeFiles/ubox.dir/runqueue.c.o CMakeFiles/ubox.dir/md5.c.o CMakeFiles/ubox.dir/kvlist.c.o CMakeFiles/ubox.dir/ulog.c.o CMakeFiles/ubox.dir/base64.c.o 
[ 97%] Linking C shared library ../../build/libcedar.so
/usr/bin/ld: /builds/zh/lede-k3/staging_dir/hostpkg/lib/libncurses.a(lib_termcap.o): relocation R_X86_64_32 against `_nc_globals' can not be used when making a shared object; recompile with -fPIC
/builds/zh/lede-k3/staging_dir/hostpkg/lib/libncurses.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
make[7]: *** [build/libcedar.so] Error 1
make[7]: Leaving directory `/ts/zh-lede-k3-ci/build_dir/hostpkg/SoftEtherVPN-5.01.9662'
make[6]: *** [src/Cedar/CMakeFiles/cedar.dir/all] Error 2
make[6]: Leaving directory `/ts/zh-lede-k3-ci/build_dir/hostpkg/SoftEtherVPN-5.01.9662'
make[5]: *** [src/hamcorebuilder/CMakeFiles/hamcorebuilder.dir/rule] Error 2
make[5]: Leaving directory `/ts/zh-lede-k3-ci/build_dir/hostpkg/SoftEtherVPN-5.01.9662'
make[4]: *** [hamcorebuilder] Error 2
make[4]: Leaving directory `/ts/zh-lede-k3-ci/build_dir/hostpkg/SoftEtherVPN-5.01.9662'
make[3]: *** [/builds/zh/lede-k3/build_dir/hostpkg/SoftEtherVPN-5.01.9662/.built] Error 2
make[3]: Leaving directory `/builds/zh/lede-k3/package/lean/softethervpn5'
time: package/lean/softethervpn5/host-compile#52.17#4.74#64.74
make[2]: *** [package/lean/softethervpn5/host/compile] Error 2
make[2]: *** Waiting for unfinished jobs....
make[6]: Leaving directory `/ts/zh-lede-k3-ci/build_dir/target-arm_cortex-a9_musl_eabi/libubox-2018-07-25-c83a84af'
[ 32%] Built target ubox
make -f CMakeFiles/blobmsg_json.dir/build.make CMakeFiles/blobmsg_json.dir/depend
make[6]: Entering directory `/ts/zh-lede-k3-ci/build_dir/target-arm_cortex-a9_musl_eabi/libubox-2018-07-25-c83a84af'
--
touch /builds/zh/lede-k3/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp/.kmod-phy-bcm-ns-usb3_installed
echo "kmod-phy-bcm-ns-usb3" >> /builds/zh/lede-k3/staging_dir/target-arm_cortex-a9_musl_eabi/pkginfo/linux.default.install
make[3]: Leaving directory `/builds/zh/lede-k3/package/kernel/linux'
time: package/kernel/linux/compile#12.85#3.75#15.64
make[2]: Leaving directory `/builds/zh/lede-k3'
make[1]: *** [/builds/zh/lede-k3/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory `/builds/zh/lede-k3'
Build failed - please re-run with -j1 to see the real error message
make: *** [world] Error 1
ERROR: Job failed: exit code 1
```

liuluo1979, 2018-11-05:
谢谢楼上帮忙，已经比我强太多了。
我只会编译到
PKG_SOURCE_DATE:=2018-10-19
PKG_SOURCE_VERSION:=cf6eef4841c59ebf15b73fe6b8c603a1c32e3edf
PKG_MIRROR_HASH:=2f5bf3d3c29e048a087a1a96a4c9661dff323aa6e9755bef31829ddf12fb1c40
从commit 3429e1bf31c16a1cd8626b074e4375eae8fe2d8a开始，我就不会编译了。

hcym, 2018-11-05:
大雕的没搞，官方的编译过，很顺利。就是开机太吃力，节点也不会弄快速的，放弃了。

hcym, 2018-11-05:
官方op应该没问题吧

solipse, 2018-11-15:
He and the official openwrt repository can not build
https://github.com/openwrt/packages/pull/6577

xinxijishuwyq, 2019-05-07:
https://github.com/Andy2244/openwrt-extra/tree/master/softethervpn5
从这里撸过来即可