coolsnowwolf/lede #767: AR71XX编译出错

ljm1447669873, 2018-11-04:
echo timestamp > s-match
g++ -fno-PIE -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings  -Wno-unused -DHAVE_CONFIG_H -I. -I. -I/home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc -I/home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/. -I/home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../include -I/home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../libcpp/include -I/home/ljm/lede/staging_dir/host/include -I/home/ljm/lede/staging_dir/host/include -I/home/ljm/lede/staging_dir/host/include  -I/home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../libdecnumber -I/home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../libdecnumber/dpd -I../libdecnumber -I/home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../libbacktrace   -o gimple-match.o -MT gimple-match.o -MMD -MP -MF ./.deps/gimple-match.TPo gimple-match.c
g++ -fno-PIE -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings  -Wno-unused -DHAVE_CONFIG_H -I. -I. -I/home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc -I/home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/. -I/home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../include -I/home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../libcpp/include -I/home/ljm/lede/staging_dir/host/include -I/home/ljm/lede/staging_dir/host/include -I/home/ljm/lede/staging_dir/host/include  -I/home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../libdecnumber -I/home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../libdecnumber/dpd -I../libdecnumber -I/home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../libbacktrace   -o generic-match.o -MT generic-match.o -MMD -MP -MF ./.deps/generic-match.TPo generic-match.c
g++ -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -DGENERATOR_FILE -fno-PIE -I. -Ibuild -I/home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc -I/home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/build -I/home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../include  -I/home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../libcpp/include  \
		-o build/genattrtab.o /home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/genattrtab.c
g++   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -DGENERATOR_FILE -fno-PIE -static-libstdc++ -static-libgcc   -o build/genattrtab \
	    build/genattrtab.o build/rtl.o build/read-rtl.o build/ggc-none.o build/vec.o build/min-insn-modes.o build/gensupport.o build/print-rtl.o build/hash-table.o build/read-md.o build/errors.o ../build-x86_64-pc-linux-gnu/libiberty/libiberty.a
build/genattrtab /home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/common.md /home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/config/mips/mips.md insn-conditions.md \
		-Atmp-attrtab.c -Dtmp-dfatab.c -Ltmp-latencytab.c
/bin/bash /home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../move-if-change tmp-attrtab.c    insn-attrtab.c
/bin/bash /home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../move-if-change tmp-dfatab.c     insn-dfatab.c
/bin/bash /home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../move-if-change tmp-latencytab.c insn-latencytab.c
echo timestamp > s-attrtab
g++ -fno-PIE -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I/home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc -I/home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/. -I/home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../include -I/home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../libcpp/include -I/home/ljm/lede/staging_dir/host/include -I/home/ljm/lede/staging_dir/host/include -I/home/ljm/lede/staging_dir/host/include  -I/home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../libdecnumber -I/home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../libdecnumber/dpd -I../libdecnumber -I/home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../libbacktrace   -o insn-attrtab.o -MT insn-attrtab.o -MMD -MP -MF ./.deps/insn-attrtab.TPo insn-attrtab.c
g++ -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -DGENERATOR_FILE -fno-PIE -I. -Ibuild -I/home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc -I/home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/build -I/home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../include  -I/home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../libcpp/include  \
		-o build/genautomata.o /home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/genautomata.c
g++   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -DGENERATOR_FILE -fno-PIE -static-libstdc++ -static-libgcc   -o build/genautomata \
	    build/genautomata.o build/rtl.o build/read-rtl.o build/ggc-none.o build/vec.o build/min-insn-modes.o build/gensupport.o build/print-rtl.o build/hash-table.o build/read-md.o build/errors.o ../build-x86_64-pc-linux-gnu/libiberty/libiberty.a -lm
build/genautomata /home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/common.md /home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/config/mips/mips.md \
	  insn-conditions.md > tmp-automata.c
/bin/bash: line 1: 10178 Killed                  build/genautomata /home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/common.md /home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/config/mips/mips.md insn-conditions.md > tmp-automata.c
make[5]: *** [s-automata] Error 137
make[5]: Leaving directory `/home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0-initial/gcc'
make[4]: *** [all-gcc] Error 2
make[4]: Leaving directory `/home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0-initial'
make[3]: *** [/home/ljm/lede/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0-initial/.built] Error 2
make[3]: Leaving directory `/home/ljm/lede/toolchain/gcc/initial'
time: toolchain/gcc/initial/compile#299.81#47.55#373.52
make[2]: *** [toolchain/gcc/initial/compile] Error 2
make[2]: Leaving directory `/home/ljm/lede'
make[1]: *** [/home/ljm/lede/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory `/home/ljm/lede'
make: *** [world] Error 2

lan542662, 2018-11-04:
你全局了吗 总归是这些原因啊。。。

ljm1447669873, 2018-11-04:
> 你全局了吗 总归是这些原因啊。。。

我在国外的vps上操作的...

hcym, 2018-11-04:
那就是环境缺失呗，ar的编译应该很多人了。我是mtk的和arm的，四五个机型，最近都正常。系统是1804实体机

ljm1447669873, 2018-11-05:
> 同样在vps上编译的，同样的错误。

换ubuntu 16.04即可

closed 2018-11-27 by coolsnowwolf