coolsnowwolf/lede #768: 11月5日编译，实时流量监测在flow加速开启时仍然无效

cindzzic, 2018-11-05:
设备：ACRH17。
固件：OpenWrt R8.1.10.24 By Lean / LuCI Master (git-18.292.64726-1262896) 。
问题：实时流量监测在flow加速开启时仍然无效。图一为开启flow加速，实时流量监测无效果，图二为关闭flow加速后，实时流量监测正常。
[image]
[image]
[内核日志.txt](https://github.com/coolsnowwolf/lede/files/2547789/default.txt)
[系统日志.txt](https://github.com/coolsnowwolf/lede/files/2547790/default.txt)

LGA1150, 2018-11-05:
开启 flow offload / sfe 会失效的原因是 wrtbwmon 使用 iptables 计数器来统计流量，而已经被 offload 的连接不会再经过 iptables
其实 OpenWrt 自带了一个不太完整的实时流量检测，在 状态-实时信息-连接里，使用了 conntrack accounting 连接计数器统计流量，因此不会受到 offload 的影响。nlbwmon 同理

cindzzic, 2018-11-05:
> 
> 
> 开启 flow offload / sfe 会失效的原因是 wrtbwmon 使用 iptables 计数器来统计流量，而已经被 offload 的连接不会再经过 iptables
> 其实 OpenWrt 自带了一个不太完整的实时流量检测，在 状态-实时信息-连接里，使用了 conntrack accounting 连接计数器统计流量，因此不会受到 offload 的影响。nlbwmon 同理

不是呀，大雕今天发布的兼容了flow。let luci app wrtbwmon compatible with Flowoffload

LGA1150, 2018-11-05:
我没看到这个 commit 做了什么实质性的修改

cindzzic, 2018-11-06:
> 
> 
> 我没看到这个 commit 做了什么实质性的修改

流量监测兼容了flow加速。

LGA1150, 2018-11-06:
这个 commit 既没有修改 flow offload 内核源码让它不绕过部分 iptables 规则，也没有修改 wrtbwmon 让它使用 conntrack accouting

closed 2018-11-27 by coolsnowwolf