coolsnowwolf/lede #771: Lean大大，apt-get 那条指令是否还适合最新版本的Ubuntu？

choushui007, 2018-11-06:
安装的时候，系统提示有几个文件是没有的，然后自动下了别的替换。

最后编译不成功，郁闷。

不知道是否与这个指令下载的文件出现了变动有关

hcym, 2018-11-06:
1804没有问题，1810还没试过

choushui007, 2018-11-06:
注意，选中 'zlib1g-dev' 而非 'libz-dev'
注意，选中 'node-uglify' 而非 'uglifyjs'
注意，选中 'git' 而非 'git-core'

这三个被替换

hcym, 2018-11-06:
升级换代变动包名而已
我是宁滥毋缺
sudo apt-get install --no-install-recommends asciidoc gcc g++ build-essential subversion git git-core libncurses5-dev zlib1g-dev gawk flex quilt libssl-dev xsltproc libxml-parser-perl mercurial bzr ecj cvs unzip fakeroot ccache build-essential  libcloog-isl-dev libreadline-dev subversion-tools xmlto libfl-dev python-launchpadlib libcurl4-gnutls-dev libexpat1-dev rsync 

兼顾老毛子编译
sudo apt-get install --no-install-recommends build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch unzip zlib1g-dev zlibc zlib-gst zlib1g build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch unzip zlib1g-dev zlibc zlib-gst zlib1g libz-dev libmbedtls-dev libpcre-ocaml libpcre++-dev gettext pkg-config libpcre3 asciidoc xmlto qemu-utils  autotools-dev libc-ares-dev libevdev-tools libsodium-dev autoconf automake debootstrap libglib2.0-dev

choushui007, 2018-11-06:
> 升级换代变动包名而已
> 我是宁滥毋缺
> sudo apt-get install --no-install-recommends asciidoc gcc g++ build-essential subversion git git-core libncurses5-dev zlib1g-dev gawk flex quilt libssl-dev xsltproc [...]

感谢

choushui007, 2018-11-08:
成功编译后，文件是不是在 lede/bin/target/ramips/mt7620 这个目录下？

allenfoxtc, 2018-11-08:
是的（我是7621）
factory: openwrt-ramips-mt7621-hc5962-squashfs-factory.bin
upgrade: openwrt-ramips-mt7621-hc5962-squashfs-sysupgrade.bin

choushui007, 2018-11-09:
已安装成功，感谢楼上两位

closed 2018-11-27 by coolsnowwolf

cyberxsboy, 2020-10-08:
> 是的（我是7621）
> factory: openwrt-ramips-mt7621-hc5962-squashfs-factory.bin
> upgrade: openwrt-ramips-mt7621-hc5962-squashfs-sysupgrade.bin

为什么我的输出目录当中只有个一个BIN文件 :  openwrt-ramips-mt7620-hiwifi_hc5761-initramfs-kernel.bin

cyberxsboy, 2020-10-08:
在Ubuntu20.04当中编译了两个固件，结果，我的输出目录当中都只有个这样一个BIN文件 : openwrt-ramips-mt7620-hiwifi_hc5761-initramfs-kernel.bin    这是不是对的呢？

cyberxsboy, 2020-10-08:
在Ubuntu20.04当中编译了两个固件，结果，我的输出目录当中都只有个这样一个BIN文件 : openwrt-ramips-mt7620-hiwifi_hc5761-initramfs-kernel.bin 这是不是对的呢？

我先前的其它版本的系统当中编译的时候，是有另外2个BIN文件的：
openwrt-ramips-mt7621-hc5962-squashfs-factory.bin
openwrt-ramips-mt7621-hc5962-squashfs-sysupgrade.bin