coolsnowwolf/lede #775: 编译完成，运行出错了？为么？

shenzt68, 2018-11-07:
/usr/lib/lua/luci/controller/guest-wifi.lua:5: attempt to call field 'loadc' (a nil value)
stack traceback:
	/usr/lib/lua/luci/controller/guest-wifi.lua:5: in function 'v'
	/usr/lib/lua/luci/dispatcher.lua:607: in function 'createtree'
	/usr/lib/lua/luci/dispatcher.lua:262: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:127: in function </usr/lib/lua/luci/dispatcher.lua:126>

shenzt68, 2018-11-07:
编译日志中发现：
cp -fpR /home/szt/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx /home/szt/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/root.orig-bcm53xx
/home/szt/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/etc/init.d/arpbind: line 5: /lib/functions.sh: No such file or directory
/home/szt/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/etc/init.d/arpbind: line 6: /lib/functions/network.sh: No such file or directory
mkdir: cannot create directory '/etc/config': Permission denied
/home/szt/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/etc/init.d/guest-wifi: line 10: uci: command not found
/home/szt/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/etc/init.d/guest-wifi: line 11: uci: command not found
/home/szt/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/etc/init.d/guest-wifi: line 12: uci: command not found
/home/szt/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/etc/init.d/guest-wifi: line 13: uci: command not found
/home/szt/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/etc/init.d/guest-wifi: line 14: uci: command not found
/home/szt/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/etc/init.d/guest-wifi: line 15: uci: command not found
/home/szt/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/etc/init.d/guest-wifi: line 16: uci: command not found
/home/szt/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/etc/init.d/guest-wifi: line 17: uci: command not found
/home/szt/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/etc/init.d/guest-wifi: line 18: uci: command not found
/home/szt/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/etc/init.d/guest-wifi: line 19: uci: command not found
/home/szt/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/etc/init.d/guest-wifi: line 20: uci: command not found
/home/szt/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/etc/init.d/guest-wifi: line 21: uci: command not found
/home/szt/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/etc/init.d/ssrpro: line 48: network_get_subnet: command not found
/home/szt/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/etc/init.d/ssrpro: line 49: network_get_ipaddr: command not found
/home/szt/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/etc/init.d/v2raypro: line 48: network_get_subnet: command not found
/home/szt/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/etc/init.d/v2raypro: line 49: network_get_ipaddr: command not found
Create '/home/szt/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/etc/filesystems'.
Add 'ntfs-3g' to known filesystems.
./etc/init.d/arpbind: line 5: /lib/functions.sh: No such file or directory
./etc/init.d/arpbind: line 6: /lib/functions/network.sh: No such file or directory
mkdir: cannot create directory '/etc/config': Permission denied
./etc/init.d/guest-wifi: line 10: uci: command not found
./etc/init.d/guest-wifi: line 11: uci: command not found
./etc/init.d/guest-wifi: line 12: uci: command not found
./etc/init.d/guest-wifi: line 13: uci: command not found
./etc/init.d/guest-wifi: line 14: uci: command not found
./etc/init.d/guest-wifi: line 15: uci: command not found
./etc/init.d/guest-wifi: line 16: uci: command not found
./etc/init.d/guest-wifi: line 17: uci: command not found
./etc/init.d/guest-wifi: line 18: uci: command not found
./etc/init.d/guest-wifi: line 19: uci: command not found

michael1981, 2018-11-09:
肯定是缺少组件，一般情况都是这样的

closed 2018-11-27 by coolsnowwolf