coolsnowwolf/lede #783: K2P CC R8.1.10.24 无法更新软件包  opkg 错误

su2uk1, 2018-11-10:
`root@OpenWrt:/tmp# opkg update
Downloading http://www.mleaf.org/downloads/chaos_calmer/15.05.1/ramips/mt7621/packages/base/Packages.gz.
Downloading http://www.mleaf.org/downloads/chaos_calmer/15.05.1/ramips/mt7621/packages/packages/Packages.gz.
Downloading http://www.mleaf.org/downloads/chaos_calmer/15.05.1/ramips/mt7621/packages/luci/Packages.gz.
Downloading http://www.mleaf.org/downloads/chaos_calmer/15.05.1/ramips/mt7621/packages/routing/Packages.gz.
Downloading http://www.mleaf.org/downloads/chaos_calmer/15.05.1/ramips/mt7621/packages/telephony/Packages.gz.
Downloading http://www.mleaf.org/downloads/chaos_calmer/15.05.1/ramips/mt7621/packages/management/Packages.gz.
Collected errors:
* opkg_download: Failed to download http://www.mleaf.org/downloads/chaos_calmer/15.05.1/ramips/mt7621/packages/base/Packages.gz, wget returned 4.
* opkg_download: Failed to download http://www.mleaf.org/downloads/chaos_calmer/15.05.1/ramips/mt7621/packages/packages/Packages.gz, wget returned 4.
* opkg_download: Failed to download http://www.mleaf.org/downloads/chaos_calmer/15.05.1/ramips/mt7621/packages/luci/Packages.gz, wget returned 4.
* opkg_download: Failed to download http://www.mleaf.org/downloads/chaos_calmer/15.05.1/ramips/mt7621/packages/routing/Packages.gz, wget returned 4.
* opkg_download: Failed to download http://www.mleaf.org/downloads/chaos_calmer/15.05.1/ramips/mt7621/packages/telephony/Packages.gz, wget returned 4.
* opkg_download: Failed to download http://www.mleaf.org/downloads/chaos_calmer/15.05.1/ramips/mt7621/packages/management/Packages.gz, wget returned 4.`

choushui007, 2018-11-10:
我怎么没看到更新的界面，难道我漏了编译哪个软件？

su2uk1, 2018-11-10:
> 我怎么没看到更新的界面，难道我漏了编译哪个软件？

直接ssh 进去 和 luci界面一样

closed 2018-11-11 by coolsnowwolf

su2uk1, 2018-11-13:
@coolsnowwolf  为森嘛关闭了 这个难道不是问题吗