coolsnowwolf/lede #794: 新版本是不是有Bug？加入V2后编译显示文件过大，去掉后重新编译也一样，7620芯片

choushui007, 2018-11-11:
echo '{  "supported_devices":["hc5661"], "version": { "dist": "OpenWrt", "version": "18.06.1", "revision": "r948-5227fd4e", "board": "ramips" } }' | fwtool -I - /home/ce6093/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-hc5661-squashfs-sysupgrade.bin
[ ! -s "/home/ce6093/lede/key-build" -o ! -s "/home/ce6093/lede/key-build.ucert" ] || { cp "/home/ce6093/lede/key-build.ucert" "/home/ce6093/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-hc5661-squashfs-sysupgrade.bin.ucert" ; usign -S -m "/home/ce6093/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-hc5661-squashfs-sysupgrade.bin" -s "/home/ce6093/lede/key-build" -x "/home/ce6093/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-hc5661-squashfs-sysupgrade.bin.sig" ; ucert -A -c "/home/ce6093/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-hc5661-squashfs-sysupgrade.bin.ucert" -x "/home/ce6093/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-hc5661-squashfs-sysupgrade.bin.sig" ; fwtool -S "/home/ce6093/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-hc5661-squashfs-sysupgrade.bin.ucert" "/home/ce6093/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-hc5661-squashfs-sysupgrade.bin" ; }
WARNING: Image file /home/ce6093/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-hc5661-squashfs-sysupgrade.bin is too big
cp /home/ce6093/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-hc5661-squashfs-sysupgrade.bin /home/ce6093/lede/bin/targets/ramips/mt7620/openwrt-ramips-mt7620-hc5661-squashfs-sysupgrade.bin
cp: cannot stat '/home/ce6093/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-ramips-mt7620-hc5661-squashfs-sysupgrade.bin': No such file or directory
Makefile:287: recipe for target '/home/ce6093/lede/bin/targets/ramips/mt7620/openwrt-ramips-mt7620-hc5661-squashfs-sysupgrade.bin' failed
make[5]: [/home/ce6093/lede/bin/targets/ramips/mt7620/openwrt-ramips-mt7620-hc5661-squashfs-sysupgrade.bin] Error 1 (ignored)
IPKG_NO_SCRIPT=1 IPKG_INSTROOT=/home/ce6093/lede/build_dir/target-mipsel_24kc_musl/root.orig-ramips TMPDIR=/home/ce6093/lede/build_dir/target-mipsel_24kc_musl/root.orig-ramips/tmp /home/ce6093/lede/staging_dir/host/bin/opkg --offline-root /home/ce6093/lede/build_dir/target-mipsel_24kc_musl/root.orig-ramips --force-postinstall --add-dest root:/ --add-arch all:100 --add-arch mipsel_24kc:200 list-installed > /home/ce6093/lede/bin/targets/ramips/mt7620/openwrt-ramips-mt7620-device-hc5661.manifest
make[5]: Leaving directory '/home/ce6093/lede/target/linux/ramips/image'

hcym, 2018-11-11:
7620肯定不行，太小，至少14M以上

coolsnowwolf, 2018-11-11:
就是过大。。

hcym, 2018-11-11:
16M可以，我的友华编译很多次，加了vpn，wireguard，zram什么的，都可以，就怕cpu扛不住

jzhou404, 2018-11-11:
把.config文件删除， 重新make menuconfig, 选择再编译试试？

choushui007, 2018-11-11:
Make menuconfig删除 $$R Plus后就正常编译了，上个版本包涵这个插件编译的时候不会报错的，奇怪

closed 2018-11-11 by coolsnowwolf