coolsnowwolf/lede #802: 使用opkg install ebtables报错

RukawaKaede, 2018-11-12:
报错信息如下: 
root@tsk3:~# opkg install ebtables kmod-ebtables-ipv6
Installing ebtables (2018-04-11-2e783b22-1) to root...
Downloading http://openwrt.proxy.ustclug.org/releases/18.06-SNAPSHOT/packages/arm_cortex-a9/base/ebtables_2018-04-11-2e783b22-1_arm_cortex-a9.ipk
Installing kmod-ebtables-ipv6 (4.14.79-1) to root...
Downloading http://openwrt.proxy.ustclug.org/releases/18.06-SNAPSHOT/targets/bcm53xx/generic/packages/kmod-ebtables-ipv6_4.14.79-1_arm_cortex-a9.ipk
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ebtables:
 * 	kernel (= 4.14.79-1-515ba6ffa9c78b51530f8fdd9e3cbeeb) *
 * opkg_install_cmd: Cannot install package ebtables.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-ebtables-ipv6:
 * 	kernel (= 4.14.79-1-515ba6ffa9c78b51530f8fdd9e3cbeeb) *
 * opkg_install_cmd: Cannot install package kmod-ebtables-ipv6.

mekaku, 2018-11-12:
内核不对，自己编译lean的固件然后在里面选择ebtables

RukawaKaede, 2018-11-12:
> 内核不对，自己编译lean的固件然后在里面选择ebtables

请问一下，可以使用windows下的WSL ubuntu（18.04）编译吗？

mekaku, 2018-11-12:
> > 内核不对，自己编译lean的固件然后在里面选择ebtables
> 
> 请问一下，可以使用windows下的WSL ubuntu（18.04）编译吗？

应该可以，之前编译了一半网速太慢就取消了，后来做了个debian双系统+ss

mekaku, 2018-11-12:
话说回来你要这个包干什么用？

RukawaKaede, 2018-11-12:
> 话说回来你要这个包干什么用？

https://github.com/coolsnowwolf/lede/issues/338
实际上是和这位朋友一样的需求，使用ipv6的，之前在其他路由器上也是使用ebtables做的。

mekaku, 2018-11-12:
> > 话说回来你要这个包干什么用？
> 
> #338
> 实际上是和这位朋友一样的需求，使用ipv6的，之前在其他路由器上也是使用ebtables做的。

你们那ipv6是dhcp-pd模式吗？是的话odhcpd ipv6 only+odhcp6c就可以，然后LAN接口这样设置
[image]

RukawaKaede, 2018-11-12:
多谢啦，具体不太清楚什么模式，我的环境是教育网ipv6，在帖子中看到过说教育网ipv6大多不支持 DHCPv6-PD。

mekaku, 2018-11-12:
你在路由器ipv6链接状态那里看一下，如果不是PD的话，那就得用另一种方式了，详见https://www.v2ex.com/t/496832?p=1#r_6295993

RukawaKaede, 2018-11-13:
根据这个链接中 https://github.com/noonien/docker-openwrt-buildroot 的Dockerfile修改成如下：

FROM ubuntu:14.04

RUN apt-get update &&\
    apt-get install -y sudo build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils wget && \
    apt-get clean

RUN useradd -m openwrt &&\
    echo 'openwrt ALL=NOPASSWD: ALL' > /etc/sudoers.d/openwrt

USER openwrt
WORKDIR /home/openwrt

RUN git clone https://github.com/coolsnowwolf/lede &&\
    openwrt/scripts/feeds update -a

不推荐使用docker -v参数挂在本地的lede目录到docker中，我这样试过，在执行./scripts/feeds update -a时报错：OpenWrt can only be built on a case-sensitive filesystem。
所以最后选择写到Dockerfile中（或者可以把Dockerfile中的最后一个RUN 命令删除，直接进入docker中手动clone）。
>