coolsnowwolf/lede #808: 编译报错，看不懂，求助

choushui007, 2018-11-13:
Applying /home/ce6093/lede/target/linux/ramips/patches-4.14/0040-nand-hack.patch using plaintext: 
patching file drivers/mtd/nand/nand_base.c
Hunk #3 FAILED at 2765.
1 out of 3 hunks FAILED -- saving rejects to file drivers/mtd/nand/nand_base.c.rej
patching file include/linux/mtd/rawnand.h
Hunk #1 FAILED at 885.
Hunk #2 succeeded at 897 (offset 4 lines).
1 out of 2 hunks FAILED -- saving rejects to file include/linux/mtd/rawnand.h.rej
Patch failed!  Please fix /home/ce6093/lede/target/linux/ramips/patches-4.14/0040-nand-hack.patch!
Makefile:26: recipe for target '/home/ce6093/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-4.14.78/.prepared_b1277342d9306c10636cd282065fd546' failed
make[4]: *** [/home/ce6093/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-4.14.78/.prepared_b1277342d9306c10636cd282065fd546] Error 1
make[4]: Leaving directory '/home/ce6093/lede/target/linux/ramips'
Makefile:13: recipe for target 'compile' failed
make[3]: *** [compile] Error 2
make[3]: Leaving directory '/home/ce6093/lede/target/linux'
time: target/linux/compile#8.00#4.59#16.57
target/Makefile:23: recipe for target 'target/linux/compile' failed
make[2]: *** [target/linux/compile] Error 2
make[2]: Leaving directory '/home/ce6093/lede'
target/Makefile:18: recipe for target '/home/ce6093/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_compile' failed
make[1]: *** [/home/ce6093/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/ce6093/lede'
/home/ce6093/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

hcym, 2018-11-13:
我靠，是不是刚才更新76驱动的？马上试试看看是不是锅

hcym, 2018-11-13:
哦，是出错，码盲就是悲哀，干脆用官方源加lean的插件编译过了，与时俱进，很好

choushui007, 2018-11-13:
> 哦，是出错，码盲就是悲哀，干脆用官方源加lean的插件编译过了，与时俱进，很好

r能否给个步骤？

hcym, 2018-11-13:
就是抓好官方的最新源码再把大佬的lean目录放到对应的(英文)包目录，其它都是正常编译了，今天的内核已经是80的，我编译的是友华7621，一切正常，v2要注意内核的什么参数一定要打开，其他地方就和官方一样

choushui007, 2018-11-14:
最新代码已经解决这个问题了

closed 2018-11-14 by choushui007