coolsnowwolf/lede #817: 谷歌连通性检查显示：连接错误，是哪里设置错误了？用18.06.1编译

shenzt68, 2018-11-14:
日志里有如下信息：
Wed Nov 14 19:35:09 2018 kern.info /usr/bin/ssr-redir[20842]: initializing ciphers... chacha20
Wed Nov 14 19:35:09 2018 kern.info /usr/bin/ssr-redir[20842]: tcp port reuse enabled
Wed Nov 14 19:35:09 2018 kern.info /usr/bin/ssr-redir[20842]: listening at 0.0.0.0:1234
Wed Nov 14 19:35:09 2018 kern.info /usr/bin/ssr-redir[20842]: running from root user
Wed Nov 14 19:35:09 2018 kern.info /usr/bin/ssr-redir[20881]: initializing ciphers... chacha20
Wed Nov 14 19:35:09 2018 kern.info /usr/bin/ssr-redir[20881]: UDP relay enabled
Wed Nov 14 19:35:09 2018 kern.info /usr/bin/ssr-redir[20881]: udp port reuse enabled
Wed Nov 14 19:35:09 2018 kern.info /usr/bin/ssr-redir[20881]: TCP relay disabled
Wed Nov 14 19:35:09 2018 kern.info /usr/bin/ssr-redir[20881]: listening at 0.0.0.0:1234
Wed Nov 14 19:35:09 2018 kern.info /usr/bin/ssr-redir[20881]: running from root user
Wed Nov 14 19:35:09 2018 daemon.info pdnsd[20897]: pdnsd-1.2.9b-par starting.
Wed Nov 14 19:35:09 2018 daemon.err pdnsd[20897]: Could not listen on tcp socket: Not a socket
Wed Nov 14 19:35:09 2018 daemon.err pdnsd[20897]: error in UDP recv: Not a socket
Wed Nov 14 19:35:09 2018 daemon.err pdnsd[20897]: error in UDP recv: Not a socket
Wed Nov 14 19:35:09 2018 daemon.err pdnsd[20897]: error in UDP recv: Not a socket
Wed Nov 14 19:35:09 2018 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Nov 14 19:35:09 2018 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Nov 14 19:35:09 2018 daemon.err pdnsd[20897]: error in UDP recv: Not a socket
Wed Nov 14 19:35:09 2018 daemon.err pdnsd[20897]: error in UDP recv: Not a socket
Wed Nov 14 19:35:09 2018 daemon.err pdnsd[20897]: error in UDP recv: Not a socket
Wed Nov 14 19:35:09 2018 daemon.err pdnsd[20897]: error in UDP recv: Not a socket
Wed Nov 14 19:35:09 2018 daemon.err pdnsd[20897]: error in UDP recv: Not a socket
Wed Nov 14 19:35:09 2018 daemon.err pdnsd[20897]: error in UDP recv: Not a socket
Wed Nov 14 19:35:09 2018 daemon.err pdnsd[20897]: error in UDP recv: Not a socket
为什么？请大屌找bug

coolsnowwolf, 2018-11-17:
端口冲突

coolsnowwolf, 2018-11-17:
5335 端口被其他程序用了

closed 2018-11-18 by coolsnowwolf

greekstreet, 2019-07-17:
@coolsnowwolf 然而并不是端口冲突，通过luci web页面启动后pdnsd会出现这个情况，但是使用/etc/init.d/shadowsocksr restart就正常了。看起来是通过web页面重启服务的话，还做了其他事情，导致了这个情况。

hcym, 2019-07-17:
时有时无，干脆重启

275761919, 2019-09-06:
问题解决了吗，我现在也遇到 luci web 操作后pdnsd就挂掉了，但是用 `pdnsd-ctl status` 查看确实正常的，必须要手动执行 `/etc/init.d/shadowsocksr restart` 或者手动 kill 掉 pdnsd 教程重启 @greekstreet

greekstreet, 2019-09-06:
@275761919 pdnsd这玩意太古老了，你稍微研究一下源码，把pdnsd换成dnsforwarder就没有问题了。也就改几个shell脚本的事情，不难。

greekstreet, 2019-09-06:
@275761919 稳妥一点的话，dnsforwader的Makefile改一下，什么启动脚本，gfwlist都不要，只要dnsforwarder本身就行了，要不然可能跟ssr plus里面的冲突。参考一下ssr plus里面pdnsd怎么启动，停止，保活，查看运行状态，多研究一下，能搞定的。

275761919, 2019-09-06:
感谢指点，我直接使用了luci-app-dnsforwarder配置成监听在5335端口，ssr-plus那边配置成使用本机的5335那个选项，就直接可以了，也不用改 shell 脚本，方便。 @greekstreet

greekstreet, 2019-09-06:
@275761919 嗯，这样也是可以的，解决问题就好。

txdywy, 2020-02-22:
重启pdnsd管用

YuefengLi, 2020-04-11:
我也遇到了一样的问题，是由于网页端重启 SSR 触发的，并且一旦出现问题，重启路由器也没啥用。

看 pdnsd staus 说 could not bind to socket，但是 netstat 出来 5335 也只有 pdnsd。杀了 pdnsd 重启它也没用。

ssh 里重启 SSR 可破。

xiaopinggao, 2022-08-19:
网络-接口-LAN-基本设置
设置DNS和网关同一个地址，不要用其他国内的自定义DNS地址。
例如
[image]

shadowsocksR采用以下配置，就解决了，而且访问外网速度很快
[image]