coolsnowwolf/lede #826: ssr-plus启动错误

dzhuqi, 2018-11-17:
昨晚pull源码编译，启动ssr-plus显示/usr/lib/lua/luci/dispatcher.lua:509: Failed to execute function dispatcher target for entry '/admin/services/shadowsocksr'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:509: Failed to execute cbi dispatcher target for entry '/admin/services/shadowsocksr/client'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/shadowsocksr/client.lua:18: bad argument #1 to 'upper' (string expected, got nil)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:509: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:127: in function </usr/lib/lua/luci/dispatcher.lua:126>

coolsnowwolf, 2018-11-17:
不要继承配置

dzhuqi, 2018-11-17:
没有保留配置，而且刷机前后特意恢复系统过也不行，在不死下刷入没有试过，今晚在试一下

closed 2018-11-17 by dzhuqi