coolsnowwolf/lede #837: 极路由1s mt7620 编译4.9内核出错

1965514404, 2018-11-18:
rm -f /home/dzc/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/hc5661-kernel.bin
cp /home/dzc/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/vmlinux /home/dzc/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/hc5661-kernel.bin
mipsel-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -I/home/dzc/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-4.9.137/arch/mips/boot/dts -I/home/dzc/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-4.9.137/arch/mips/boot/dts/include -I/home/dzc/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-4.9.137/include/ -undef -D__DTS__  -o /home/dzc/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/hc5661-kernel.bin.dtb.tmp ../dts/HC5661.dts
/home/dzc/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-4.9.137/scripts/dtc/dtc -O dtb -i../dts/  -o /home/dzc/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/hc5661-kernel.bin.dtb /home/dzc/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/hc5661-kernel.bin.dtb.tmp
rm -f /home/dzc/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/hc5661-kernel.bin.dtb.tmp
/home/dzc/lede/staging_dir/host/bin/patch-dtb /home/dzc/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/hc5661-kernel.bin /home/dzc/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/hc5661-kernel.bin.dtb
DT size used is default of 16KB
DTB marker not found!
make[5]: *** [/home/dzc/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/hc5661-kernel.bin] Error 255
make[5]: Leaving directory `/home/dzc/lede/target/linux/ramips/image'
make[4]: *** [install] Error 2
make[4]: Leaving directory `/home/dzc/lede/target/linux/ramips'
make[3]: *** [install] Error 2
make[3]: Leaving directory `/home/dzc/lede/target/linux'
time: target/linux/install#810.67#78.11#902.33
make[2]: *** [target/linux/install] Error 2
make[2]: Leaving directory `/home/dzc/lede'
make[1]: *** [/home/dzc/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory `/home/dzc/lede'
make: *** [world] 错误 2
[dzc@ice lede]$

choushui007, 2018-11-19:
编译通过，但是bin文件缩小了3M，现在才12M左右，奇怪

1965514404, 2018-11-19:
> 编译通过，但是bin文件缩小了3M，现在才12M左右，奇怪

因为驱动少了几个,我发现的有rt2800soc,mt76,sfe,等
少前两个会没有无线驱动

closed 2018-11-27 by coolsnowwolf