coolsnowwolf/lede #871: 求教：Netgear Wndr4300v1 全新刷机但是就是无法启动V2RAY

yhl10000, 2018-11-30:
我自己编译了 coolsnowwolf/lede 固件，menuconfig加上了v2ray，编译到固件，但是全新安装后，配置pppoe后，可以上网，但是无论是直接启用v2ray，还是配置了服务器的v2ray，都是**V2ray 未运行**。尝试了若干天，一直无法解决。求助。

附上我的menu config， 请大神帮忙看看。

[coolsnowwolf/lede menuconfig file](https://gist.github.com/yhl10000/8be4063df6dd9af559fe15c3c30a1cf7)

lyin888, 2018-11-30:
开启fpu浮点
make kernel_menuconfig
kwrnel type
选中 mips fpu emulator

yhl10000, 2018-11-30:
我试试，但是ar7xxx/ar9xxx的也要开浮点么？

xros, 2020-05-14:
@yhl10000 我想问一下。4300v1的效率怎么样？卡吗？能跑多快。v2测速的时候，cpu，内存使用情况。

yhl10000, 2020-05-15:
> @yhl10000 我想问一下。4300v1的效率怎么样？卡吗？能跑多快。v2测速的时候，cpu，内存使用情况。

我没有安装其他的插件，我的是200M的宽带，平时用没问题，油管也流畅。

xros, 2020-05-16:
> > @yhl10000 我想问一下。4300v1的效率怎么样？卡吗？能跑多快。v2测速的时候，cpu，内存使用情况。
> 
> @xros 我没有安装其他的插件，我的是200M的宽带，平时用没问题，油管也流畅。

openwrt 能跑满200mbps带宽？我刷的openwrt，没有硬件nat。带宽最大才150mbps。上不去了。

yhl10000, 2020-05-18:
> > > @yhl10000 我想问一下。4300v1的效率怎么样？卡吗？能跑多快。v2测速的时候，cpu，内存使用情况。
> > 
> > 
> > @xros 我没有安装其他的插件，我的是200M的宽带，平时用没问题，油管也流畅。
> 
> openwrt 能跑满200mbps带宽？我刷的openwrt，没有硬件nat。带宽最大才150mbps。上不去了。

@xros 我有这样的情形，但我不确定是否是openwrt的问题导致带宽上不去。虽然运营商给了200M，但这也受限于相关设备的，比如网线，路由器等。