coolsnowwolf/lede #881: 不显示S-SR-plus

yfeercom, 2018-12-05:
12月4日编译完成后 服务里面不显示ssr,xiaomimini和wndr4300V1 都不显示

zcluo, 2018-12-05:
我的也是虚拟机菜单入口没了

mymail6, 2018-12-05:
New3 和WR1200JS 也都没显示 ssr plus luci界面

choushui007, 2018-12-05:
极路由1S也不显示，我以为是编译的时候没加上，然后删除 .config。 make manuconfig的时候出错。。。

lyin888, 2018-12-05:
在推上看到以为推油转发的需要在命令行输入一个命令来解密，命令没来得及保存哪就给删了

lyin888, 2018-12-05:
删除 package/lean/luci-app-ssr-plus/luasrc/controller/shadowsocksr.lua文件，然后替换，
现在shadowsocksr.lua被加密了

zcluo, 2018-12-05:
这是要收费的节奏吗

ypdenet, 2018-12-05:
我以为我眼瞎 没勾选呢

hcym, 2018-12-05:
这么有什么意义呢？害得我用小山源码编译了老老版ssr，加密格式太少，又改了服务端的配置，居然比新版本快一点点。滑稽查局

lyin888, 2018-12-05:
不是收费，天朝你懂得，安全为好

lm317379829, 2018-12-05:
echo  0xDEADBEEF > /etc/config/google_fu_mode

lyin888, 2018-12-05:
感谢楼上的

zcluo, 2018-12-05:
要怎么解密？

shenzt68, 2018-12-05:
echo 0xDEADBEEF > /etc/config/google_fu_mode
这个能有用？不是老毛子才这么干的吗？

choushui007, 2018-12-06:
> echo 0xDEADBEEF > /etc/config/google_fu_mode
> 这个能有用？不是老毛子才这么干的吗？

亲测有用，理解一下人家，上面亚历山大

coolsnowwolf, 2018-12-06:
最近风声紧，作为彩蛋吧~

dongfangy2011, 2018-12-07:
> > echo 0xDEADBEEF> / etc / config / google_fu_mode 
> > 这个能有用？不是老毛子才这么干的吗？
> 
> 亲测有用，理解一下人家，上面亚历山大

请问： 在哪里输入这个命令呢？直接在路由器里还是再编译环境的Ubuntu里。谢谢

choushui007, 2018-12-07:
> > > echo 0xDEADBEEF> / etc / config / google_fu_mode
> > > 这个能有用？不是老毛子才这么干的吗？
> > 
> > 
> > 亲测有用，理解一下人家，上面亚历山大
> 
> 请问： 在哪里输入这个命令呢？直接在路由器里还是再编译环境的Ubuntu里。谢谢

电脑端SSH登录路由器，然后照着打即可

choushui007, 2018-12-07:
> 最近风声紧，作为彩蛋吧~

能否同步一下 openwrt 项目的变动？

lucky0l, 2018-12-07:
还有这操作，好吧，这个得记录一下，以防止后面忘了又找不到……

MxCen, 2018-12-07:
马克一下。。。。

snowrice, 2018-12-08:
怪不得明明編譯了卻看不見，原來要念咒語才能看到

ixxxxxx, 2018-12-10:
弱弱的问下，`echo 0xDEADBEEF > /etc/config/google_fu_mode`开启后不能添加ss服务器？
是哪里没有设置好吗？
[image]

wxlost, 2018-12-10:
原来是彩蛋...我以为是我编译问题...编译了一天刷了30多次....我...

lm317379829, 2018-12-11:
> 弱弱的问下，`echo 0xDEADBEEF > /etc/config/google_fu_mode`开启后不能添加ss服务器？
> 是哪里没有设置好吗？
> [image]

编译的时候没有选shadowsocks New Version

ixxxxxx, 2018-12-11:
> > 弱弱的问下，`echo 0xDEADBEEF > /etc/config/google_fu_mode`开启后不能添加ss服务器？
> > 是哪里没有设置好吗？
> > [image]
> 
> 编译的时候没有选shadowsocks New Version

貌似是这样的，感谢。

rabbired, 2018-12-17:
原来是彩蛋，MAKE一下。

waybq, 2018-12-20:
原来是这样啊，谢谢，今天编译了几次，也刷了几次，都没菜单，但是容量是增加的。原来是这样啊。
谢谢各位了。

RicardoKlose, 2018-12-20:
居然还有彩蛋，mark一下

a1768623020, 2018-12-21:
我以为我眼睛瞎了呢。。。原来是彩蛋！谢谢大佬。

closed 2018-12-22 by coolsnowwolf

lan542662, 2018-12-31:
@wxlost 你比我还人才。刷30多次。。。有问题找男神。enjoy~

yunnan0317, 2019-02-24:
纳闷了半天...找着根了

whlsxl, 2019-05-25:
git checkout -b xx 2915c44a11ca0ee40b51ff5d9c18a0d
Love & Peace!!!

mgsqz, 2019-07-04:
Mark一下

xlbljz, 2019-07-04:
原来如此，感谢各位

TheProjectM, 2019-07-22:
Easter egg

qq280948982, 2020-03-18:
> echo 0xDEADBEEF > /etc/config/google_fu_mode

输入了彩蛋命令，但是界面上依然没有显示，怎么破？

jzjohn789, 2020-04-30:
输入了，但是依然没有，是删除了？

kktt007, 2020-05-01:
> 输入了，但是依然没有，是删除了？

helloworld

jzjohn789, 2020-05-01:
> > 输入了，但是依然没有，是删除了？
> 
> helloworld

???什么意思

xlbljz, 2020-05-01:
> 输入了，但是依然没有，是删除了？

排一下安装的错误吧

jzjohn789, 2020-05-01:
> > 输入了，但是依然没有，是删除了？
> 
> 排一下安装的错误吧

没错的，我现在想装USB打印服务器，一直弄不好，知道怎么弄吗，要重新编译吗

xlbljz, 2020-05-01:
实在弄不好可以去论坛找现成的固件刷

xlbljz, 2020-05-01:
> > > 输入了，但是依然没有，是删除了？
> > 
> > 
> > 排一下安装的错误吧
> 
> 没错的，我现在想装USB打印服务器，一直弄不好，知道怎么弄吗，要重新编译吗

实在弄不好可以去论坛找现成的固件刷

jzjohn789, 2020-05-01:
> > > > 输入了，但是依然没有，是删除了？
> > > 
> > > 
> > > 排一下安装的错误吧
> > 
> > 
> > 没错的，我现在想装USB打印服务器，一直弄不好，知道怎么弄吗，要重新编译吗
> 
> 实在弄不好可以去论坛找现成的固件刷

就是找不到。。。。。。搞通宵了

xlbljz, 2020-05-01:
> > > > > 输入了，但是依然没有，是删除了？
> > > > 
> > > > 
> > > > 排一下安装的错误吧
> > > 
> > > 
> > > 没错的，我现在想装USB打印服务器，一直弄不好，知道怎么弄吗，要重新编译吗
> > 
> > 
> > 实在弄不好可以去论坛找现成的固件刷
> 
> 就是找不到。。。。。。搞通宵了

恩山无线论坛，上面有很多大神做的固件。

jzjohn789, 2020-05-01:
> 恩山无线论坛

看看吧

jzjohn789, 2020-05-01:
> > > > > > 输入了，但是依然没有，是删除了？
> > > > > 
> > > > > 
> > > > > 排一下安装的错误吧
> > > > 
> > > > 
> > > > 没错的，我现在想装USB打印服务器，一直弄不好，知道怎么弄吗，要重新编译吗
> > > 
> > > 
> > > 实在弄不好可以去论坛找现成的固件刷
> > 
> > 
> > 就是找不到。。。。。。搞通宵了
> 
> 恩山无线论坛，上面有很多大神做的固件。

不是说他们弄了一堆后门吗？

xlbljz, 2020-05-01:
> 不是说他们弄了一堆后门吗？

用人不疑，疑人不用。自己衡量。

jzjohn789, 2020-05-01:
> > 不是说他们弄了一堆后门吗？
> 
> 用人不疑，疑人不用。自己衡量。

教我一下自编译吧