coolsnowwolf/lede #911: 无法挂载usb设备, 没有 /dev/sdx, scsi_mod 未加载

Ljqiii, 2018-12-15:
内核日志:

[    0.000000] Linux version 4.14.87 (lean@laptopljq) (gcc version 7.3.0 (OpenWrt GCC 7.3.0 r1009-36e8cbe6)) #0 SMP Wed Dec 12 08:09:34 2018
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Xiaomi Mi Router 3G
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 10000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] On node 0 totalpages: 65536
[    0.000000] free_area_init_node: node 0, pgdat 8052fce0, node_mem_map 81003000
[    0.000000]   Normal zone: 512 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000] random: get_random_bytes called from 0x80533740 with crng_init=0
[    0.000000] percpu: Embedded 14 pages/cpu @81210000 s26704 r8192 d22448 u57344
[    0.000000] pcpu-alloc: s26704 r8192 d22448 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 65024
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Writing ErrCtl register=00000160
[    0.000000] Readback ErrCtl register=00000160
[    0.000000] Memory: 253768K/262144K available (4516K kernel code, 231K rwdata, 564K rodata, 244K init, 244K bss, 8376K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 256
[    0.000000] CPU Clock: 880MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.000010] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
[    0.007807] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.073976] pid_max: default: 32768 minimum: 301
[    0.078744] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.085251] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.094047] Hierarchical SRCU implementation.
[    0.099196] smp: Bringing up secondary CPUs ...
[    9.353823] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    9.353833] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    9.353845] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    9.353982] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.163946] Synchronize counters for CPU 1: done.
[    9.380864] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    9.380872] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    9.380880] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    9.380951] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.255035] Synchronize counters for CPU 2: done.
[    9.470159] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    9.470166] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    9.470173] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    9.470250] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.340214] Synchronize counters for CPU 3: done.
[    0.370069] smp: Brought up 1 node, 4 CPUs
[    0.377841] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.387628] futex hash table entries: 1024 (order: 3, 32768 bytes)
[    0.393925] pinctrl core: initialized pinctrl subsystem
[... 1225 lines cut ...]
[  487.970870] [<8044a6ec>] 0x8044a6ec
[  487.974342] [<80010184>] 0x80010184
[  487.977827] [<80451564>] 0x80451564
[  487.981301] [<8000d0a4>] 0x8000d0a4
[  487.984773] [<8000d0b4>] 0x8000d0b4
[  487.988261] [<8009d804>] 0x8009d804
[  487.991748] [<80057b68>] 0x80057b68
[  487.995222] [<800155b0>] 0x800155b0
[  487.998708] [<80071684>] 0x80071684
[  488.002186] [<800717b8>] 0x800717b8
[  488.005662] [<8007185c>] 0x8007185c
[  488.009134] [<80315910>] 0x80315910
[  488.012607] [<80050694>] 0x80050694
[  488.016081] [<80075af0>] 0x80075af0
[  488.019553] [<80052b78>] 0x80052b78
[  488.023028] [<80070b20>] 0x80070b20
[  488.026500] [<80070b20>] 0x80070b20
[  488.029972] [<8024ac54>] 0x8024ac54
[  488.033446] [<8024ab00>] 0x8024ab00
[  488.036920] [<8024acc0>] 0x8024acc0
[  488.040393] [<80070b20>] 0x80070b20
[  488.043867] [<80468c8c>] 0x80468c8c
[  488.047342] [<80249ca8>] 0x80249ca8
[  488.050839] [<8000b508>] 0x8000b508
[  488.054322]

RukawaKaede, 2018-12-16:
同出现问题：
```
[ 1263.590144] usb 1-1: new full-speed USB device number 10 using ohci-platform
[ 1263.810106] usb 1-1: device descriptor read/64, error -62
[ 1264.140110] usb 1-1: device descriptor read/64, error -62
[ 1264.470136] usb 1-1: new full-speed USB device number 11 using ohci-platform
[ 1264.690111] usb 1-1: device descriptor read/64, error -62
[ 1265.020139] usb 1-1: device descriptor read/64, error -62
[ 1265.140149] usb usb1-port1: attempt power cycle
[ 1265.690119] usb 1-1: new full-speed USB device number 12 using ohci-platform
[ 1266.130117] usb 1-1: device not accepting address 12, error -62
[ 1266.340200] usb 1-1: new full-speed USB device number 13 using ohci-platform
[ 1266.780116] usb 1-1: device not accepting address 13, error -62
[ 1266.786100] usb usb1-port1: unable to enumerate USB device
```

xuansongzhang, 2018-12-18:
最新版本的有同样的问题

closed 2018-12-22 by coolsnowwolf