coolsnowwolf/lede #921: 18号编译的K3固件 搜不到5G

TobyTong, 2018-12-17:
今天的K3固件编译完找不到5G

hcym, 2018-12-18:
昨天改了吧，大佬看来不准备支持k3了

hcym, 2018-12-18:
那就不清楚了，最近k3的v2必须替换文件才能启动，所以一直处于老的版本，编译其它的7800，新3什么的都不错

heartofrainbow, 2018-12-18:
同样的问题，18号编译的最新K3固件无5G信号，编译是删掉文件夹从头再来的，刷固件时是不保存配置的，现在贴上日志：
`Mon Dec 17 22:50:35 2018 daemon.notice netifd: radio1 (24375): command failed: No error information (-524)
Mon Dec 17 22:50:35 2018 user.notice mac80211: Failed command: iw dev wlan1 del
Mon Dec 17 22:50:35 2018 daemon.notice netifd: radio1 (24500): command failed: No error information (-524)
Mon Dec 17 22:50:35 2018 user.notice mac80211: Failed command: iw dev wlan1 del
Mon Dec 17 22:50:35 2018 user.notice mac80211: Failed command: iw phy phy1 set antenna all all
Mon Dec 17 22:50:35 2018 daemon.notice netifd: radio1 (24500): command failed: Not supported (-95)
Mon Dec 17 22:50:35 2018 user.notice mac80211: Failed command: iw phy phy1 set antenna_gain 0
Mon Dec 17 22:50:35 2018 daemon.notice netifd: radio1 (24500): command failed: I/O error (-5)
Mon Dec 17 22:50:35 2018 user.notice mac80211: Failed command: iw phy phy1 set distance 0
Mon Dec 17 22:50:35 2018 daemon.notice netifd: radio1 (24500): command failed: Too many open files in system (-23)
Mon Dec 17 22:50:35 2018 user.notice mac80211: Failed command: iw phy phy1 interface add wlan1 type __ap
Mon Dec 17 22:50:35 2018 daemon.err hostapd: Configuration file: /var/run/hostapd-phy1.conf
Mon Dec 17 22:50:35 2018 kern.info kernel: [  564.316008] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
Mon Dec 17 22:50:35 2018 kern.info kernel: [  564.328470] br-lan: port 3(wlan1) entered blocking state
Mon Dec 17 22:50:35 2018 kern.info kernel: [  564.333896] br-lan: port 3(wlan1) entered disabled state
Mon Dec 17 22:50:35 2018 kern.info kernel: [  564.339461] device wlan1 entered promiscuous mode
Mon Dec 17 22:50:35 2018 daemon.notice hostapd: wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
Mon Dec 17 22:50:35 2018 daemon.notice hostapd: wlan1: interface state COUNTRY_UPDATE->HT_SCAN
Mon Dec 17 22:50:35 2018 daemon.err hostapd: Could not set channel for kernel driver
Mon Dec 17 22:50:35 2018 daemon.err hostapd: Interface initialization failed
Mon Dec 17 22:50:35 2018 daemon.notice hostapd: wlan1: interface state HT_SCAN->DISABLED
Mon Dec 17 22:50:35 2018 daemon.notice hostapd: wlan1: AP-DISABLED
Mon Dec 17 22:50:35 2018 daemon.notice hostapd: wlan1: interface state DISABLED->DISABLED
Mon Dec 17 22:50:35 2018 daemon.notice hostapd: wlan1: AP-DISABLED
Mon Dec 17 22:50:35 2018 daemon.notice hostapd: wlan1: CTRL-EVENT-TERMINATING
Mon Dec 17 22:50:35 2018 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan1 wasn't started
Mon Dec 17 22:50:35 2018 daemon.notice hostapd: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
Mon Dec 17 22:50:36 2018 kern.info kernel: [  564.642204] device wlan1 left promiscuous mode
Mon Dec 17 22:50:36 2018 kern.info kernel: [  564.646748] br-lan: port 3(wlan1) entered disabled state
Mon Dec 17 22:50:36 2018 daemon.warn odhcpd[2236]: A default route is present but there is no public prefix on br-lan thus we don't announce a default route!
Mon Dec 17 22:50:36 2018 daemon.notice hostapd: ELOOP: remaining socket: sock=21 eloop_data=0xb6e8c790 user_data=0 handler=0x40838
Mon Dec 17 22:50:36 2018 daemon.notice netifd: radio1 (24500): cat: can't open '/var/run/wifi-phy1.pid': No such file or directory
Mon Dec 17 22:50:36 2018 daemon.notice netifd: radio1 (24500): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path (/proc/exe)
Mon Dec 17 22:50:36 2018 daemon.notice netifd: radio1 (24500): Command failed: Invalid argument
Mon Dec 17 22:50:36 2018 kern.info kernel: [  565.248853] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready`

上面的日志是尝试修改5G wifi时的输出。

同时在无线页面中，所有的传输速率均为0或？，5G wifi的模式一直显示为client，加密显示为“-”

TobyTong, 2018-12-18:
> 那就不清楚了，最近k3的v2必须替换文件才能启动，所以一直处于老的版本，编译其它的7800，新3什么的都不错

40MHZ可以的    80和160搜不到5G  模式也只有40是master。。。先放弃了

mekaku, 2018-12-18:
没看到commits么 把wifi去了

hcym, 2018-12-18:
都正常了，今天大雕改了就正常运行了，v2也正常运行了，呵呵，谢谢大雕谢谢cctv

closed 2018-12-20 by coolsnowwolf