coolsnowwolf/lede #924: 编译错误

yunmi323, 2018-12-18:
-install/usr/lib/samba/*.so* /home/coolmo/LEDE/lede/build_dir/target-mipsel_24kc_musl/samba-4.9.2/ipkg-mipsel_24kc/samba4-libs/usr/lib/
find /home/coolmo/LEDE/lede/build_dir/target-mipsel_24kc_musl/samba-4.9.2/ipkg-mipsel_24kc/samba4-libs -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package samba4-libs is missing dependencies for the following libraries:
libcap.so.2
Makefile:394: recipe for target '/home/coolmo/LEDE/lede/bin/packages/mipsel_24kc/base/samba4-libs_4.9.2-2_mipsel_24kc.ipk' failed
make[3]: *** [/home/coolmo/LEDE/lede/bin/packages/mipsel_24kc/base/samba4-libs_4.9.2-2_mipsel_24kc.ipk] Error 1
make[3]: Leaving directory '/home/coolmo/LEDE/lede/package/lean/samba4'
time: package/lean/samba4/compile#0.62#0.73#1.85
package/Makefile:107: recipe for target 'package/lean/samba4/compile' failed
make[2]: *** [package/lean/samba4/compile] Error 2
make[2]: Leaving directory '/home/coolmo/LEDE/lede'
package/Makefile:103: recipe for target '/home/coolmo/LEDE/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/coolmo/LEDE/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/coolmo/LEDE/lede'
/home/coolmo/LEDE/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

换了好多梯子都一样是出错，怎么搞

realsteam, 2018-12-18:
我也是这个问题

hzjnet, 2018-12-19:
看：
Package samba4-libs is missing dependencies for the following libraries:
libcap.so.2
Makefile:394: recipe for target '/home/coolmo/LEDE/lede/bin/packages/mipsel_24kc/base/samba4-libs_4.9.2-2_mipsel_24kc.ipk' failed

你需要取消samba4的相关选项

yunmi323, 2018-12-19:
不选Samba4也是有同样的错误，libcap2版本也是最新的，真无解

forsan-co, 2018-12-19:
好像不选samba4需要对luci关联的包也取消，才能编译成功，最简单的做法就是重新选择一个硬件版本，保存退出，再重新make menuconfig去选择你需要的。

yunmi323, 2018-12-19:
这方法简单好用

closed 2018-12-20 by coolsnowwolf