coolsnowwolf/lede #934: automount的问题

HiGarfield, 2018-12-20:
按照现有自动挂载脚本，挂载多个盘的话，重启之后可能出现顺序错乱，导致某些应用的路径出问题。

推荐的改法：

09-automount（必须小于10）

#!/bin/sh

set_fstab(){
	mkdir -p /var/lock
	lock /var/lock/fstab.lck
	my_fstype="`block info | grep "/dev/$device" | awk -F 'TYPE="' '{print $2}' | sed 's/\"//'`"

	[ -n "$my_fstype" ] && {
		logger -t Auto-Mount "New block.File system:${my_fstype}"
		[ "$my_fstype" = 'swap' ] && {

			n=$(uci show fstab | grep "fstab.@swap" | grep -c "=swap")

			[ $n -gt 0 ] && {
				for i in $(seq 0 $n); do
					old_swap="$(uci get fstab.@swap[$i].device)"
					[ "$old_swap" = "/dev/$device" ] && {
						FLAG="SKIP"
						break
					}
				done
			}

			[ "$FLAG" != "SKIP" ] && {
				uci add fstab swap
				uci set fstab.@swap[$n]="swap"
				uci set fstab.@swap[$n].enabled=1
				uci set fstab.@swap[$n].device="/dev/$device"
				uci commit fstab
			}
		} || {	
			my_mnt_opts=
			my_supported_flag="1"
			case "$my_fstype" in
				'ext2'|'ext3'|'ext4')
					my_mnt_opts="noatime" ;;
				'ntfs')
					[ $(lsmod | grep -c ufsd) -ge 1 ] && {
						my_fstype="ufsd"
						my_mnt_opts="noatime,nls=utf8,force"
					} || {
						my_fstype="ntfs-3g"
						my_mnt_opts="noatime,iocharset=utf8,big_writes"
					} ;;
				'exfat')
					my_mnt_opts="noatime" ;;
				'vfat')
					my_mnt_opts="iocharset=utf8,umask=0000,dmask=0000,fmask=0000" ;;
				*)
					my_supported_flag="0"
					logger -t Auto-Mount "File system:${my_fstype} not supported"
					;;
			esac

			[ "$my_supported_flag" == "1" ] && {
				uci add fstab mount
				uci set fstab.@mount[-1]="mount"
				uci set fstab.@mount[-1].enabled=1
				uci set fstab.@mount[-1].uuid="$get_uuid"
				uci set fstab.@mount[-1].target="/mnt/$device"
				uci set fstab.@mount[-1].fstype="$my_fstype"
				[ -n "$my_mnt_opts" ] && uci set fstab.@mount[-1].options="$my_mnt_opts"
				uci commit fstab
			}
		}		
	}
	lock -u /var/lock/fstab.lck
}

blkdev=`dirname $DEVPATH`
[ `basename $blkdev` != "block" ] && {
	device=`basename $DEVPATH`
	case "$ACTION" in
		add)
			get_uuid=`block info | grep "/dev/${device}" | awk -F "UUID=" '{print $2}'| awk -F "\"" '{print $2}'`
			[ -n "$get_uuid" ] && {
				have_uuid=$(uci show fstab | grep -c "$get_uuid")
				[ "$have_uuid" = "0" ] && set_fstab
				have_uuid=$(uci show fstab | grep -c "$get_uuid")
				[ "$have_uuid" != "0" ] && {
					logger -t Auto-Mount "Block /dev/${device} added."
					logger -t Auto-Mount "UUID=$get_uuid"
					mkdir -p /mnt/$device
				}
			}
		;;
		remove)
			mountpoint=`mount | grep /dev/$device | sed 's/.* on \(.*\) type.*/\1/'`
			[ -n "$mountpoint" ] && {
				/bin/force-unmount "$mountpoint" >/dev/null 2>&1
			}
		;;
	esac
}

以上脚本中，/bin/force-unmount的内容为：

#!/bin/sh

function stop_occupying_service()
{
	local dir_name process_line process_name service_name
	dir_name="$1"
	lsof | grep "$dir_name" | awk '{print $2}' | uniq | while read process_line
	do
		process_name=$(basename "$process_line")
		grep -w -l "$process_name" /etc/init.d/* | uniq | while read service_name
		do
			"$service_name" stop
		done
	done
}

function kill_occupying_process()
{
	local dir_name pids
	dir_name="$1"
	pids=`lsof | grep "$dir_name" | awk '{print $1}' | uniq | awk BEGIN{RS=EOF}'{gsub(/\n/," ");print}'`
	[ -n "$pids" ] && kill -9 $pids >/dev/null 2>&1
}

[ $# -ne 1 ] && exit 1
mountpoint="$1"
stop_occupying_service "$mountpoint"
kill_occupying_process "$mountpoint"
umount -l "$mountpoint"
exit 0

注：busybox需要选上lsof命令。

HiGarfield, 2018-12-20:
注意，上面的代码显示有问题。反引号被github吃掉了。。。

edmondfr, 2018-12-21:
期待大雕更新

closed 2018-12-22 by coolsnowwolf