coolsnowwolf/lede #935: K3编译的时候，去掉了 SSPLUS，添加了V2ray，编译出错。

mrclock, 2018-12-21:
双清过的。
log
----------------------------
make[3]: Entering directory '/root/lede/package/lean/v2ray'
rm -rf /root/lede/build_dir/target-arm_cortex-a9_musl_eabi/v2ray-v4.9.0/.pkgdir/v2ray.installed /root/lede/build_dir/target-arm_cortex-a9_musl_eabi/v2ray-v4.9.0/.pkgdir/v2ray
mkdir -p /root/lede/build_dir/target-arm_cortex-a9_musl_eabi/v2ray-v4.9.0/.pkgdir/v2ray
install -d -m0755 /root/lede/build_dir/target-arm_cortex-a9_musl_eabi/v2ray-v4.9.0/.pkgdir/v2ray/usr/bin/v2ray
install -m0755 /root/lede/build_dir/target-arm_cortex-a9_musl_eabi/v2ray-v4.9.0/ /root/lede/build_dir/target-arm_cortex-a9_musl_eabi/v2ray-v4.9.0/.pkgdir/v2ray/usr/bin/v2ray/v2ray
install: omitting directory '/root/lede/build_dir/target-arm_cortex-a9_musl_eabi/v2ray-v4.9.0/'
Makefile:121: recipe for target '/root/lede/build_dir/target-arm_cortex-a9_musl_eabi/v2ray-v4.9.0/.pkgdir/v2ray.installed' failed
make[3]: *** [/root/lede/build_dir/target-arm_cortex-a9_musl_eabi/v2ray-v4.9.0/.pkgdir/v2ray.installed] Error 1
make[3]: Leaving directory '/root/lede/package/lean/v2ray'
time: package/lean/v2ray/compile#0.16#0.16#0.36
package/Makefile:107: recipe for target 'package/lean/v2ray/compile' failed
make[2]: *** [package/lean/v2ray/compile] Error 2

hcym, 2018-12-21:
v2包下载太慢，浏览器下载快的很。

mrclock, 2018-12-21:
大雕说 不支持高通了。。。自己搞。。。关issue.

closed 2018-12-21 by mrclock