coolsnowwolf/lede #940: R7800WiFi缓慢？

lijun0326, 2018-12-21:
当前配置：
```
config wifi-device 'radio0'
        option type 'mac80211'
        option hwmode '11a'
        option path 'soc/1b500000.pci/pci0000:00/0000:00:00.0/0000:01:00.0'
        option country 'US'
        option htmode 'VHT80'
        option channel '157'
        option legacy_rates '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid '**REMOVED**'
        option encryption 'psk2+ccmp'
        option key '**REMOVED**'
        option wpa_disable_eapol_key_retries '1'
```
在隔壁openwrt论坛里面讨论了下，上游应该是没有问题的，不知道这里的源做了什么修改导致只有20M/s下载和40M/s上传
https://forum.openwrt.org/t/netgear-r7800-5g-wifi-quite-slow/27278

lijun0326, 2018-12-21:
> 你要确认无线驱动能力大小最好用内网传输测试比较靠谱

当然是拿局域网NAS测的。。

hcym, 2018-12-22:
160用官方可以的，或者那个xr500固件，我刷了除了管理界面慢，没办法ssr，其它感觉没什么区别。手上都是433的手机。

laincat, 2018-12-22:
> 
> 
> 160用官方可以的，或者那个xr500固件，我刷了除了管理界面慢，没办法ssr，其它感觉没什么区别。手上都是433的手机。

怎么设置？可以发一份配置看看吗？需要什么软件包？

lijun0326, 2018-12-22:
@laincat 应该是直接在WiFi里面就有160MHz频宽的选项

lijun0326, 2018-12-22:
@hcym ssr那些直接把这个项目的package拷过去一起编译就好

closed 2018-12-22 by coolsnowwolf

lijun0326, 2018-12-22:
直接close了还行