coolsnowwolf/lede #944: 树莓派编译错误

sun-live, 2018-12-23:
近期发现三个朋友都在编译树莓派版本时临近尾声报错

> recipe for target '/home/forsan/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-brcm2708_bcm2709/tmp/openwrt-brcm2708-bcm2709-rpi-2-squashfs-factory.img.gz' failed

与朋友折腾后发现
与官方源对比，menuconfig生成的.config文件内在如下位置发生缺失，：

```
 #
 # Image Options
 #
 CONFIG_BRCM2708_SD_BOOT_PARTSIZE=20
 #
```

目前手动于此处插入目标分区大小，如下：
```
#
# Image Options
#
CONFIG_BRCM2708_SD_BOOT_PARTSIZE=20
CONFIG_TARGET_ROOTFS_PARTSIZE=256
#
```

即可解决, thanks~

kangxicome, 2018-12-24:
nod :) same problem.

forsan-co, 2018-12-26:
真心感谢 @sun-live 的回复，上面的issue就是我提的，但是这个手动添加在第二次编译的时候，config文件会被覆盖掉，就会丢掉
#
# Image Options
#
CONFIG_BRCM2708_SD_BOOT_PARTSIZE=20
CONFIG_TARGET_ROOTFS_PARTSIZE=256
#
我用另外的办法可以编译成功（我的编译的是2b的树莓派，）：
在target image 里面，勾选ext4，就出现root filesystem partition size的选项，我修改成了你上面的256以后，config文件出现了
CONFIG_TARGET_EXT4_BLOCKSIZE=4096
CONFIG_TARGET_EXT4_BLOCKSIZE_4K=y
CONFIG_TARGET_EXT4_RESERVED_PCT=0
CONFIG_TARGET_ROOTFS_EXT4FS=y
CONFIG_TARGET_ROOTFS_PARTSIZE=256
并且编译成功。

kangxicome, 2018-12-26:
ext4在3和3b下也可以，主要是没找到改arch里面的哪个配置文件可以直接出现这个选项

forsan-co, 2018-12-28:
今天git pull后又报错了，等会儿排查下看看

forsan-co, 2018-12-28:
新版本变异报错，猜测原因是linux内核升级造成的
`lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-brcm2708_bcm2709/fixup_x.dat: No such file or directory`
文件start_x.elf也找不到，所以编译错误，将https://github.com/raspberrypi/firmware 中boot里面的文件放进
`lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-brcm2708_bcm2709/`
这个目录后编译成功，还没有刷写，手上的TF读卡器坏了...
推测是kernel: bump 4.14 to 4.14.90 这个升级造成的，里面有许多brcm2708或者brcm2708_bcm2709的变化, 我只是小白所以暂且也不知道错误原因，还得大屌排查

coolsnowwolf, 2018-12-28:
已修复，rm .config 重新选择即可

closed 2019-01-02 by coolsnowwolf