coolsnowwolf/lede #947: 请求frp增加apm821xx平台的支持

palmerxiaobiao, 2018-12-24:
#
# Copyright (C) 2015-2016 OpenWrt.org
#
# This is free software, licensed under the GNU General Public License v3.
#

include $(TOPDIR)/rules.mk

PKG_NAME:=frpc
PKG_VERSION:=0.21.0
PKG_RELEASE:=1

ifeq ($(ARCH),mipsel)
	FRPC_ARCH:=mipsle
endif
ifeq ($(ARCH),mips)
	FRPC_ARCH:=mips
endif
ifeq ($(ARCH),i386)
	FRPC_ARCH:=386
endif
ifeq ($(ARCH),x86_64)
	FRPC_ARCH:=amd64
endif
ifeq ($(ARCH),arm)
	FRPC_ARCH:=arm
endif
ifeq ($(ARCH),arm64)
	FRPC_ARCH:=arm64
endif

PKG_LICENSE:=Apache-2.0
多谢大神开源，在使用源码编译网件wndr4700的固件时，frp编译出错，希望大神可以增加对apm821xx平台的支持。

LGA1150, 2018-12-30:
试试 #966

palmerxiaobiao, 2018-12-31:
ng_dir/target-powerpc_464fp_musl/lib -L/home/openwrt/lede/staging_dir/toolchain-powerpc_464fp_gcc-7.3.0_musl/usr/lib -L/home/openwrt/lede/staging_dir/toolchain-powerpc_464fp_gcc-7.3.0_musl/lib -znow -zrelro " make  -C /home/openwrt/lede/build_dir/target-powerpc_464fp_musl/frp-0.22.0/. AR="powerpc-openwrt-linux-musl-gcc-ar" AS="powerpc-openwrt-linux-musl-gcc -c -Os -pipe -mcpu=464fp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/openwrt/lede/build_dir/target-powerpc_464fp_musl/frp-0.22.0:frp-0.22.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=powerpc-openwrt-linux-musl-ld NM="powerpc-openwrt-linux-musl-gcc-nm" CC="powerpc-openwrt-linux-musl-gcc" GCC="powerpc-openwrt-linux-musl-gcc" CXX="powerpc-openwrt-linux-musl-g++" RANLIB="powerpc-openwrt-linux-musl-gcc-ranlib" STRIP=powerpc-openwrt-linux-musl-strip OBJCOPY=powerpc-openwrt-linux-musl-objcopy OBJDUMP=powerpc-openwrt-linux-musl-objdump SIZE=powerpc-openwrt-linux-musl-size CROSS="powerpc-openwrt-linux-musl-" ARCH="powerpc" ;
make[4]: Entering directory '/home/openwrt/lede/build_dir/target-powerpc_464fp_musl/frp-0.22.0'
go fmt ./...
go build -o bin/frps ./cmd/frps
cmd/frps/root.go:24:2: cannot find package "github.com/fatedier/frp/g" in any of:
        /home/openwrt/lede/staging_dir/hostpkg/lib/go-cross/src/github.com/fatedier/frp/g (from $GOROOT)
        /home/openwrt/go/src/github.com/fatedier/frp/g (from $GOPATH)
cmd/frps/root.go:25:2: cannot find package "github.com/fatedier/frp/models/config" in any of:
        /home/openwrt/lede/staging_dir/hostpkg/lib/go-cross/src/github.com/fatedier/frp/models/config (from $GOROOT)
        /home/openwrt/go/src/github.com/fatedier/frp/models/config (from $GOPATH)
cmd/frps/root.go:26:2: cannot find package "github.com/fatedier/frp/server" in any of:
        /home/openwrt/lede/staging_dir/hostpkg/lib/go-cross/src/github.com/fatedier/frp/server (from $GOROOT)
        /home/openwrt/go/src/github.com/fatedier/frp/server (from $GOPATH)
cmd/frps/root.go:27:2: cannot find package "github.com/fatedier/frp/utils/log" in any of:
        /home/openwrt/lede/staging_dir/hostpkg/lib/go-cross/src/github.com/fatedier/frp/utils/log (from $GOROOT)
        /home/openwrt/go/src/github.com/fatedier/frp/utils/log (from $GOPATH)
cmd/frps/root.go:28:2: cannot find package "github.com/fatedier/frp/utils/util" in any of:
        /home/openwrt/lede/staging_dir/hostpkg/lib/go-cross/src/github.com/fatedier/frp/utils/util (from $GOROOT)
        /home/openwrt/go/src/github.com/fatedier/frp/utils/util (from $GOPATH)
cmd/frps/root.go:29:2: cannot find package "github.com/fatedier/frp/utils/version" in any of:
        /home/openwrt/lede/staging_dir/hostpkg/lib/go-cross/src/github.com/fatedier/frp/utils/version (from $GOROOT)
        /home/openwrt/go/src/github.com/fatedier/frp/utils/version (from $GOPATH)
cmd/frps/main.go:18:2: cannot find package "github.com/fatedier/golib/crypto" in any of:
        /home/openwrt/lede/staging_dir/hostpkg/lib/go-cross/src/github.com/fatedier/golib/crypto (from $GOROOT)
        /home/openwrt/go/src/github.com/fatedier/golib/crypto (from $GOPATH)
cmd/frps/root.go:22:2: cannot find package "github.com/spf13/cobra" in any of:
        /home/openwrt/lede/staging_dir/hostpkg/lib/go-cross/src/github.com/spf13/cobra (from $GOROOT)
        /home/openwrt/go/src/github.com/spf13/cobra (from $GOPATH)
Makefile:20: recipe for target 'frps' failed
make[4]: *** [frps] Error 1
make[4]: Leaving directory '/home/openwrt/lede/build_dir/target-powerpc_464fp_musl/frp-0.22.0'
Makefile:45: recipe for target '/home/openwrt/lede/build_dir/target-powerpc_464fp_musl/frp-0.22.0/.built' failed
make[3]: *** [/home/openwrt/lede/build_dir/target-powerpc_464fp_musl/frp-0.22.0/.built] Error 2
make[3]: Leaving directory '/home/openwrt/lede/package/lean/frpc'
time: package/lean/frpc/compile#0.28#0.03#0.58
package/Makefile:107: recipe for target 'package/lean/frpc/compile' failed
make[2]: *** [package/lean/frpc/compile] Error 2
make[2]: Leaving directory '/home/openwrt/lede'
package/Makefile:103: recipe for target '/home/openwrt/lede/staging_dir/target-powerpc_464fp_musl/stamp/.package_compile' failed
make[1]: *** [/home/openwrt/lede/staging_dir/target-powerpc_464fp_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/openwrt/lede'
/home/openwrt/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
openwrt@openmediavault:~/lede$
谢谢LAG1150的回复，修改后编译没法通过。

closed 2019-01-02 by coolsnowwolf

AnxQ, 2020-09-11:
告诉我一下是不是不支持这个架构 让我死心