coolsnowwolf/lede #948: v2ray : x509: certificate signed by unknown authority

wwng2333, 2018-12-25:
2018/12/25 12:07:30 [Warning] failed to handler mux client connection > v2ray.com/core/proxy/vmess/outbound: failed to find an available destination > v2ray.com/core/common/retry: [v2ray.com/core/transport/internet/websocket: failed to dial WebSocket > v2ray.com/core/transport/internet/websocket: failed to dial to (ws://*.*.*/downloads/):  > x509: certificate signed by unknown authority] > v2ray.com/core/common/retry: all retry attempts failed

在PC上一切正常，怀疑缺少根证书

附上证书链：
[image]

matthew728960, 2018-12-25:
试试打开 "允许不安全的证书" 选项

wwng2333, 2018-12-31:
80d6d03b49273bec16a0da30141b85e0debfffd2

closed 2018-12-31 by wwng2333