coolsnowwolf/lede #953: 极二HC5761编译出来没有5G wifi,是不支持吗？

W666692, 2018-12-27:
极二HC5761编译出来没有5G wifi,是不支持吗？

W666692, 2018-12-27:
MT7620

LGA1150, 2018-12-28:
用官方的 dts 编译，并选上 kmod-mt76x0e

W666692, 2018-12-28:
哪里有？不太懂！能具体点吗？

W666692, 2018-12-28:
是要替换lede/target/linux/ramips/dts/HC5761.dts这个文件的内容吗？

W666692, 2018-12-28:
/dts-v1/;

#include "HC5X61.dtsi"

/ {
	compatible = "hiwifi,hc5761", "hiwifi,hc5x61", "ralink,mt7620a-soc";
	model = "HiWiFi HC5761";

	aliases {
		led-boot = &led_system;
		led-failsafe = &led_system;
		led-running = &led_system;
		led-upgrade = &led_system;
	};

	gpio-leds {
		compatible = "gpio-leds";

		led_system: system {
			label = "hc5761:blue:system";
			gpios = <&gpio0 9 GPIO_ACTIVE_LOW>;
		};

		internet {
			label = "hc5761:blue:internet";
			gpios = <&gpio0 11 GPIO_ACTIVE_LOW>;
		};

		wlan2g {
			label = "hc5761:blue:wlan2g";
			gpios = <&gpio3 0 GPIO_ACTIVE_LOW>;
		};

		wlan5g {
			label = "hc5761:blue:wlan5g";
			gpios = <&gpio0 7 GPIO_ACTIVE_LOW>;
		};
	};
};

&ehci {
	status = "okay";
};

&ohci {
	status = "okay";
};

&pcie {
	status = "okay";
};

&pcie0 {
	mt76@0,0 {
		reg = <0x0000 0 0 0 0>;
		mediatek,mtd-eeprom = <&factory 0x8000>;
		ieee80211-freq-limit = <5000000 6000000>;
	};
};

W666692, 2018-12-28:
这个吗？

LGA1150, 2018-12-28:
是

W666692, 2018-12-28:
修改这一个文件内容就好了吗？这个是编译配置吗？真不懂，一头雾水

LGA1150, 2018-12-28:
别忘了选上 kmod-mt76x0e ，在 kernel modules - wireless drivers 里

W666692, 2018-12-28:
我试下，谢谢

W666692, 2018-12-29:
编译出来有5G了，指示灯也不行，放弃了

LGA1150, 2018-12-29:
你要去 LED 设置页手动改一下

W666692, 2018-12-29:
改了没用，netdev没办法触发，灯不亮

LGA1150, 2018-12-31:
netdev bug 好像是最近才出现的

W666692, 2018-12-31:
应该是更新了内核的问题

closed 2019-01-02 by coolsnowwolf