coolsnowwolf/lede #971: 最新的frp0.22编译出错

jxph1123, 2018-12-31:
can't load package: package github.com/fatedier/frp/cmd/frpc: cannot find package "github.com/fatedier/frp/cmd/frpc" in any of:
	/home/etek/lede/staging_dir/hostpkg/lib/go-cross/src/github.com/fatedier/frp/cmd/frpc (from $GOROOT)
	/home/etek/lede/build_dir/target-mipsel_24kc_musl/frpc-0.22.0/.go_work/build/src/github.com/fatedier/frp/cmd/frpc (from $GOPATH)
can't load package: package .: no Go files in /home/etek/lede/build_dir/target-mipsel_24kc_musl/frpc-0.22.0/.go_work/build
Makefile:46: recipe for target '/home/etek/lede/build_dir/target-mipsel_24kc_musl/frpc-0.22.0/.built' failed
make[3]: *** [/home/etek/lede/build_dir/target-mipsel_24kc_musl/frpc-0.22.0/.built] Error 1
make[3]: Leaving directory '/home/etek/lede/package/lean/frpc'
time: package/lean/frpc/compile#0.22#0.28#0.63
package/Makefile:107: recipe for target 'package/lean/frpc/compile' failed
make[2]: *** [package/lean/frpc/compile] Error 2
make[2]: Leaving directory '/home/etek/lede'
package/Makefile:103: recipe for target '/home/etek/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/etek/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/etek/lede'
/home/etek/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

LGA1150, 2018-12-31:
这个 Makefile 放到官方的 buildroot 是可以正常编译的
你试试把 Makefile 中所有的 frpc 替换为 frp

LGA1150, 2018-12-31:
```
make[1]: Entering directory '/root/openwrt'
make[2]: Entering directory '/root/openwrt/package/libs/toolchain'
make[2]: Entering directory '/root/openwrt/feeds/packages/lang/golang/golang'
make[2]: Leaving directory '/root/openwrt/feeds/packages/lang/golang/golang'
time: package/feeds/packages/golang/host-compile#0.15#0.08#0.25
echo "libc" >> /root/openwrt/staging_dir/target-mipsel_24kc_musl/pkginfo/toolchain.default.install
echo "libgcc" >> /root/openwrt/staging_dir/target-mipsel_24kc_musl/pkginfo/toolchain.default.install
echo "libstdcpp" >> /root/openwrt/staging_dir/target-mipsel_24kc_musl/pkginfo/toolchain.default.install
echo "libpthread" >> /root/openwrt/staging_dir/target-mipsel_24kc_musl/pkginfo/toolchain.default.install
echo "librt" >> /root/openwrt/staging_dir/target-mipsel_24kc_musl/pkginfo/toolchain.default.install
make[2]: Leaving directory '/root/openwrt/package/libs/toolchain'
time: package/libs/toolchain/compile#0.09#0.02#0.39
make[2]: Entering directory '/hdd/lede/package/lean/frpc'
touch /root/openwrt/build_dir/target-mipsel_24kc_musl/frp-0.22.0/.prepared_020e824d4935d03e702a5cd82154fc96_6664517399ebbbc92a37c5bb081b5c53_check
. /root/openwrt/include/shell.sh; gzip -dc /root/openwrt/dl/frp-0.22.0.tar.gz | tar -C /root/openwrt/build_dir/target-mipsel_24kc_musl/frp-0.22.0/.. -xf - 
[ ! -d ./src/ ] || cp -fpR ./src/. /root/openwrt/build_dir/target-mipsel_24kc_musl/frp-0.22.0
touch /root/openwrt/build_dir/target-mipsel_24kc_musl/frp-0.22.0/.prepared_020e824d4935d03e702a5cd82154fc96_6664517399ebbbc92a37c5bb081b5c53
rm -f /root/openwrt/build_dir/target-mipsel_24kc_musl/frp-0.22.0/.configured_*
rm -f /root/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.frpc_installed
( cd /root/openwrt/build_dir/target-mipsel_24kc_musl/frp-0.22.0 ; mkdir -p /root/openwrt/build_dir/target-mipsel_24kc_musl/frp-0.22.0/.go_work/build/bin /root/openwrt/build_dir/target-mipsel_24kc_musl/frp-0.22.0/.go_work/build/src /root/openwrt/build_dir/target-mipsel_24kc_musl/frp-0.22.0/.go_work/cache /root/openwrt/build_dir/target-mipsel_24kc_musl/frp-0.22.0/.go_work/tmp ; files=$(find ./ -type d -a \( -path './.git' -o -path './.go_work' \) -prune -o \! -type d -print | sed 's|^\./||') ; if [ "" != 1 ]; then code=$(echo "$files" | grep '\.\(c\|cc\|go\|h\|hh\|proto\|s\)$') ; testdata=$(echo "$files" | grep '\(^\|/\)testdata/') ; for pattern in ; do extra=$(echo "$extra"; echo "$files" | grep "$pattern") ; done ; files=$(echo "$code"; echo "$testdata"; echo "$extra") ; files=$(echo "$files" | grep -v '^[[:space:]]*$' | sort -u) ; fi ; echo "Copying files from /root/openwrt/build_dir/target-mipsel_24kc_musl/frp-0.22.0 into /root/openwrt/build_dir/target-mipsel_24kc_musl/frp-0.22.0/.go_work/build/src/github.com/fatedier/frp" ; for file in $files; do echo $file ; dest=/root/openwrt/build_dir/target-mipsel_24kc_musl/frp-0.22.0/.go_work/build/src/github.com/fatedier/frp/$file ; mkdir -p $(dirname $dest) ; cp -fpR $file $dest ; done ; echo ; link_contents() { local src=$1 ; local dest=$2 ; local dirs dir base ; if [ -n "$(find $src -mindepth 1 -maxdepth 1 -name '*.go' \! -type d)" ]; then echo "$src is already a Go library" ; return 1 ; fi ; dirs=$(find $src -mindepth 1 -maxdepth 1 -type d) ; for dir in $dirs; do base=$(basename $dir) ; if [ -d $dest/$base ]; then case $dir in */usr/share/gocode/src/github.com/fatedier/frp) echo "github.com/fatedier/frp is already installed. Please check for circular dependencies." ;; *) link_contents $src/$base $dest/$base ;; esac ; else echo "...${src#/root/openwrt/staging_dir/target-mipsel_24kc_musl/usr/share/gocode/src}/$base" ; ln -sf $src/$base $dest/$base ; fi ; done ; } ; if [ "" != 1 ]; then if [ -d /root/openwrt/staging_dir/target-mipsel_24kc_musl/usr/share/gocode/src ]; then echo "Symlinking directories from /root/openwrt/staging_dir/target-mipsel_24kc_musl/usr/share/gocode/src into /root/openwrt/build_dir/target-mipsel_24kc_musl/frp-0.22.0/.go_work/build/src" ; link_contents /root/openwrt/staging_dir/target-mipsel_24kc_musl/usr/share/gocode/src /root/openwrt/build_dir/target-mipsel_24kc_musl/frp-0.22.0/.go_work/build/src ; else echo "/root/openwrt/staging_dir/target-mipsel_24kc_musl/usr/share/gocode/src does not exist, skipping symlinks" ; fi ; else echo "Not building binaries, skipping symlinks" ; fi ; echo ; )
Copying files from /root/openwrt/build_dir/target-mipsel_24kc_musl/frp-0.22.0 into /root/openwrt/build_dir/target-mipsel_24kc_musl/frp-0.22.0/.go_work/build/src/github.com/fatedier/frp
assets/assets.go
assets/statik/statik.go
client/admin.go
client/admin_api.go
client/control.go
client/event/event.go
client/health/health.go
client/proxy/proxy.go
client/proxy/proxy_manager.go
client/proxy/proxy_wrapper.go
client/service.go
client/visitor.go
client/visitor_manager.go
cmd/frpc/main.go
cmd/frpc/sub/http.go
cmd/frpc/sub/https.go
cmd/frpc/sub/reload.go
cmd/frpc/sub/root.go
cmd/frpc/sub/status.go
cmd/frpc/sub/stcp.go
cmd/frpc/sub/tcp.go
cmd/frpc/sub/udp.go
cmd/frpc/sub/xtcp.go
cmd/frps/main.go
cmd/frps/root.go
g/g.go
models/config/client_common.go
models/config/proxy.go
models/config/server_common.go
models/config/visitor.go
models/consts/consts.go
models/errors/errors.go
models/msg/ctl.go
models/msg/msg.go
models/plugin/http_proxy.go
models/plugin/plugin.go
models/plugin/socks5.go
models/plugin/static_file.go
models/plugin/unix_domain_socket.go
models/proto/udp/udp.go
models/proto/udp/udp_test.go
server/control.go
server/dashboard.go
server/dashboard_api.go
[... cut ...]
if [ -d /root/openwrt/tmp/stage-frpc ]; then (cd /root/openwrt/tmp/stage-frpc; find ./ > /root/openwrt/tmp/stage-frpc.files); 	SHELL= flock /root/openwrt/tmp/.staging-dir.flock -c ' mv /root/openwrt/tmp/stage-frpc.files /root/openwrt/staging_dir/target-mipsel_24kc_musl/packages/frpc.list && cp -fpR /root/openwrt/tmp/stage-frpc/* /root/openwrt/staging_dir/target-mipsel_24kc_musl/; '; fi
rm -rf /root/openwrt/tmp/stage-frpc
touch /root/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.frpc_installed
make[2]: Leaving directory '/hdd/lede/package/lean/frpc'
time: package/lean/frpc/compile#23.34#1.66#10.90
make[1]: Leaving directory '/root/openwrt'
```

closed 2019-01-02 by coolsnowwolf

tefiszx, 2019-01-04:
错误提示貌似是没有下载源码到本地造成的。楼主看看本地源码是否存在。

LGA1150, 2019-01-04:
@tefiszx 我在官方 buildroot 上编译是通过的
这个分支太乱了，没有及时、正确地同步上游，是时候重置一下了