coolsnowwolf/lede #977: 内核版本不一致安装不上 NFS

Hentioe, 2019-01-02:
安装过程：
````
root@OpenWrt:/data# opkg install kmod-vxlan
Installing kmod-vxlan (4.14.63-1) to root...
Downloading http://openwrt.proxy.ustclug.org/releases/18.06.1/targets/ipq40xx/generic/packages/kmod-vxlan_4.14.63-1_arm_cortex-a7_neon-vfpv4.ipk
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-vxlan:
 * 	kernel (= 4.14.63-1-55f569c94f99f14d0eaf9e39ac0ed4b4) * 	kernel (= 4.14.63-1-55f569c94f99f14d0eaf9e39ac0ed4b4) * 	kernel (= 4.14.63-1-55f569c94f99f14d0eaf9e39ac0ed4b4) * 
 * opkg_install_cmd: Cannot install package kmod-vxlan.
````

系统内核版本：
````
root@OpenWrt:/data# uname -a
Linux OpenWrt 4.14.82 #0 SMP Mon Dec 10 12:36:30 2018 armv7l GNU/Linux
````

我没有修改过软件源，路由器是 Asus RT-ACRH17。我在 menuconfig 的菜单里边找遍了也没有看到这个包（或其它跟 NFS 相关的包），我该如何安装 NFS？

closed 2019-01-03 by coolsnowwolf

RukawaKaede, 2019-01-04:
make menuconfig -->File System里头有NFS

Hentioe, 2019-01-04:
@RukawaKaede 嗯，是的，我后来通过 “/” 键搜索找到了。不过还是感谢回复～
顺便我这边编译的时候，一个 nfs 相关的包（不是 nfs-kernel-server）通不过，手动改 Makefile 才成功编译，修改方式跟这里一样：https://github.com/openwrt/packages/commit/b1be3f9c06d5286e1e698edb580e629ae03f0393

在 `TARGET_CFLAGS` 添加 `-Wno-error=format-security` 参数，写在这里希望有同样问题的能搜到～

RukawaKaede, 2019-01-04:
@Hentioe 嗯嗯，👍，我也被这个内核不一致问题搞得头大，没办法才自己编译固件去，踩了不少坑23333

justlovemaki, 2019-05-20:
各位大哥，编译的时候加了nfs的相关package，但是nfs，nfsd命令，服务都不存在，这个怎么处理啊？