coolsnowwolf/lede #978: 代码更新到9.1.1后编译出错，mtdsplit

monsterzzzz, 2019-01-03:
CC      drivers/mtd/mtdsplit/mtdsplit_uimage.o
drivers/mtd/mtdsplit/mtdsplit_uimage.c:244:34: error: array type has incomplete element type 'struct of_device_id'
 static const struct of_device_id mtdsplit_uimage_of_match_table[] = {
                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
drivers/mtd/mtdsplit/mtdsplit_uimage.c:245:4: error: field name not in record or union initializer
  { .compatible = "denx,uimage" },
    ^
drivers/mtd/mtdsplit/mtdsplit_uimage.c:245:4: note: (near initialization for 'mtdsplit_uimage_of_match_table')
drivers/mtd/mtdsplit/mtdsplit_uimage.c:311:34: error: array type has incomplete element type 'struct of_device_id'
 static const struct of_device_id mtdsplit_uimage_netgear_of_match_table[] = {
                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
drivers/mtd/mtdsplit/mtdsplit_uimage.c:312:4: error: field name not in record or union initializer
  { .compatible = "netgear,uimage" },
    ^
drivers/mtd/mtdsplit/mtdsplit_uimage.c:312:4: note: (near initialization for 'mtdsplit_uimage_netgear_of_match_table')
drivers/mtd/mtdsplit/mtdsplit_uimage.c:311:34: warning: 'mtdsplit_uimage_netgear_of_match_table' defined but not used [-Wunused-variable]
 static const struct of_device_id mtdsplit_uimage_netgear_of_match_table[] = {
                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
drivers/mtd/mtdsplit/mtdsplit_uimage.c:244:34: warning: 'mtdsplit_uimage_of_match_table' defined but not used [-Wunused-variable]
 static const struct of_device_id mtdsplit_uimage_of_match_table[] = {
                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
scripts/Makefile.build:304: recipe for target 'drivers/mtd/mtdsplit/mtdsplit_uimage.o' failed
make[8]: *** [drivers/mtd/mtdsplit/mtdsplit_uimage.o] Error 1
scripts/Makefile.build:555: recipe for target 'drivers/mtd/mtdsplit' failed
make[7]: *** [drivers/mtd/mtdsplit] Error 2
scripts/Makefile.build:555: recipe for target 'drivers/mtd' failed
make[6]: *** [drivers/mtd] Error 2
Makefile:1026: recipe for target 'drivers' failed
make[5]: *** [drivers] Error 2

Lancenas, 2019-01-03:
WNDR4300 v1 更新编译也出现主题错误

hcym, 2019-01-03:
动作大，还是十一月份的稳妥

choushui007, 2019-01-03:
一样编译出错

closed 2019-01-03 by coolsnowwolf

tefiszx, 2019-01-05:
我刚刚编译4300 v1 成功没出错。是不是profile选错了？