coolsnowwolf/lede #986: HC5962 2.4g

ddvcx, 2019-01-05:


coolsnowwolf, 2019-01-05:
日志看看

ddvcx, 2019-01-06:
Sun Jan  6 08:04:29 2019 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 34:13:e8:99:4e:31
Sun Jan  6 08:04:34 2019 cron.err crond[1164]: time disparity of 2432 minutes detected
Sun Jan  6 08:05:29 2019 daemon.info hostapd: wlan1: STA 34:13:e8:99:4e:31 IEEE 802.11: authenticated
Sun Jan  6 08:05:29 2019 daemon.info hostapd: wlan1: STA 34:13:e8:99:4e:31 IEEE 802.11: associated (aid 1)
Sun Jan  6 08:05:29 2019 daemon.notice hostapd: wlan1: STA-OPMODE-SMPS-MODE-CHANGED 34:13:e8:99:4e:31 off
Sun Jan  6 08:05:29 2019 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 34:13:e8:99:4e:31
Sun Jan  6 08:05:29 2019 daemon.info hostapd: wlan1: STA 34:13:e8:99:4e:31 WPA: pairwise key handshake completed (RSN)
Sun Jan  6 08:05:36 2019 daemon.err uhttpd[1191]: luci: accepted login on / for root from 192.168.2.215
Sun Jan  6 08:05:50 2019 kern.info kernel: [  243.744633] device wlan0 left promiscuous mode
Sun Jan  6 08:05:50 2019 kern.info kernel: [  243.749311] br-lan: port 3(wlan0) entered disabled state
Sun Jan  6 08:05:50 2019 daemon.notice hostapd: wlan0: interface state ENABLED->DISABLED
Sun Jan  6 08:05:50 2019 daemon.notice hostapd: wlan0: AP-DISABLED
Sun Jan  6 08:05:50 2019 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Sun Jan  6 08:05:50 2019 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Sun Jan  6 08:05:50 2019 daemon.notice hostapd: nl80211: Failed to remove interface wlan0 from bridge br-lan: Invalid argument
Sun Jan  6 08:05:51 2019 user.notice mac80211: Failed command: iw phy phy0 set antenna all all
Sun Jan  6 08:05:59 2019 daemon.notice netifd: radio0 (5517): command failed: No such device (-19)
Sun Jan  6 08:05:59 2019 user.notice mac80211: Failed command: iw dev tmp.radio0 del
Sun Jan  6 08:05:59 2019 user.notice mac80211: Failed command: iw phy phy0 set antenna all all
Sun Jan  6 08:05:59 2019 daemon.err hostapd: Configuration file: /var/run/hostapd-phy0.conf
Sun Jan  6 08:06:00 2019 kern.info kernel: [  254.073725] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Sun Jan  6 08:06:02 2019 kern.info kernel: [  255.485319] br-lan: port 3(wlan0) entered blocking state
Sun Jan  6 08:06:02 2019 kern.info kernel: [  255.490659] br-lan: port 3(wlan0) entered disabled state
Sun Jan  6 08:06:02 2019 kern.info kernel: [  255.496517] device wlan0 entered promiscuous mode
Sun Jan  6 08:06:02 2019 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Sun Jan  6 08:06:02 2019 daemon.err hostapd: Using interface wlan0 with hwaddr fc:fc:43:26:60:60 and ssid "DX"
Sun Jan  6 08:06:02 2019 kern.info kernel: [  256.331156] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Sun Jan  6 08:06:02 2019 kern.info kernel: [  256.337812] br-lan: port 3(wlan0) entered blocking state
Sun Jan  6 08:06:02 2019 kern.info kernel: [  256.343206] br-lan: port 3(wlan0) entered forwarding state
Sun Jan  6 08:06:02 2019 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->ENABLED
Sun Jan  6 08:06:02 2019 daemon.notice hostapd: wlan0: AP-ENABLED
Sun Jan  6 08:06:02 2019 daemon.notice netifd: Network device 'wlan0' link is up

daka2017, 2019-01-09:
我的没有问题啊

ddvcx, 2019-01-10:
试了只选路由器芯片型号直接编译，还是有这个问题
另外也试了恩山坛友编译的古剑，也是有这个问题
https://www.right.com.cn/FORUM/forum.php?mod=viewthread&tid=399426&highlight=b70
https://www.right.com.cn/FORUM/forum.php?mod=viewthread&tid=368253&highlight=b70

coolsnowwolf, 2019-01-11:
已修复

ddvcx, 2019-01-14:
> 已修复

11月12日得代码编译估计，还是用不了2.4G

Mon Jan 14 09:04:26 2019 kern.info kernel: [  313.874776] device wlan0 left promiscuous mode
Mon Jan 14 09:04:26 2019 kern.info kernel: [  313.879376] br-lan: port 2(wlan0) entered disabled state
Mon Jan 14 09:04:26 2019 daemon.notice hostapd: wlan0: interface state ENABLED->DISABLED
Mon Jan 14 09:04:26 2019 daemon.notice hostapd: wlan0: AP-DISABLED
Mon Jan 14 09:04:26 2019 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Mon Jan 14 09:04:26 2019 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Mon Jan 14 09:04:26 2019 daemon.notice hostapd: nl80211: Failed to remove interface wlan0 from bridge br-lan: Invalid argument
Mon Jan 14 09:04:27 2019 user.notice mac80211: Failed command: iw phy phy0 set antenna all all
Mon Jan 14 09:04:36 2019 daemon.notice netifd: radio0 (19223): command failed: No such device (-19)
Mon Jan 14 09:04:36 2019 user.notice mac80211: Failed command: iw dev tmp.radio0 del
Mon Jan 14 09:04:36 2019 user.notice mac80211: Failed command: iw phy phy0 set antenna all all
Mon Jan 14 09:04:38 2019 daemon.err hostapd: Configuration file: /var/run/hostapd-phy0.conf
Mon Jan 14 09:04:39 2019 kern.info kernel: [  326.597331] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Mon Jan 14 09:04:39 2019 kern.info kernel: [  326.606459] br-lan: port 2(wlan0) entered blocking state
Mon Jan 14 09:04:39 2019 kern.info kernel: [  326.611820] br-lan: port 2(wlan0) entered disabled state
Mon Jan 14 09:04:39 2019 kern.info kernel: [  326.617491] device wlan0 entered promiscuous mode
Mon Jan 14 09:04:39 2019 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Mon Jan 14 09:04:39 2019 daemon.err hostapd: Using interface wlan0 with hwaddr fc:fc:43:26:60:60 and ssid "D"
Mon Jan 14 09:04:39 2019 kern.info kernel: [  327.378290] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Mon Jan 14 09:04:39 2019 kern.info kernel: [  327.384893] br-lan: port 2(wlan0) entered blocking state
Mon Jan 14 09:04:39 2019 kern.info kernel: [  327.390182] br-lan: port 2(wlan0) entered forwarding state
Mon Jan 14 09:04:39 2019 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->ENABLED
Mon Jan 14 09:04:39 2019 daemon.notice hostapd: wlan0: AP-ENABLED
Mon Jan 14 09:04:39 2019 daemon.notice netifd: Network device 'wlan0' link is up

closed 2019-01-14 by coolsnowwolf