coolsnowwolf/lede #988: 1月4号R9.1.1的V2Ray启动不了

L-vivi, 2019-01-05:
Sat Jan  5 22:11:06 2019 user.notice shadowsocksr: ssr redir tcp error.restart!
Sat Jan  5 22:11:07 2019 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sat Jan  5 22:11:07 2019 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sat Jan  5 22:11:11 2019 cron.info crond[18295]: crond (busybox 1.30.0) started, log level 8
Sat Jan  5 22:11:12 2019 daemon.err pdnsd[18455]: Could not bind tcp socket: Address in use
Sat Jan  5 22:11:12 2019 daemon.err pdnsd[18455]: Could not bind to udp socket: Address in use
Sat Jan  5 22:11:12 2019 daemon.err pdnsd[18455]: tcp and udp initialization failed. Exiting.
Sat Jan  5 22:11:13 2019 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sat Jan  5 22:11:13 2019 user.notice dnsmasq: Allowing 127.0.0.0/8 responses

hcym, 2019-01-05:
什么型号，mtk的大雕忘了什么fup浮点，其它的应该正常使用，我是编译了新3，7800，ac58u都ok

coolsnowwolf, 2019-01-05:
已修复了

L-vivi, 2019-01-06:
谢谢大雕，但是mips好像没修复，MT7621新3路由还是不能启动

hcym, 2019-01-06:
看看内核选项，浮点运算有没有，没有就自己选择。新3编译的v2正常使用

L-vivi, 2019-01-07:
真的可以用了，我忘记删除.config编译了，重新make menuconfig现在正常了，谢谢大雕

closed 2019-01-08 by coolsnowwolf