coolsnowwolf/lede #999: v2ray的插件可否增加订阅功能

kangxicome, 2019-01-09:
ks的v2ray也可以通过vmess直接订阅多个服务器，希望后续可加入...

WKZD, 2019-02-19:
同求，订阅增加支持v2ray和ss

ntgeralt, 2019-03-01:
44个Issues， 楼主忙不过来了……

lingzhi2262071123, 2019-03-26:
订阅来一发

alliswell2day, 2019-04-04:
我自己改修了，订阅增加支持v2ray和ss。 支持 这个魔改 https://github.com/rico93/ss-panel-v3-mod_Uim 面
[image]

jiushibushuo, 2019-04-26:
> 我自己改修了，订阅增加支持v2ray和ss。 支持 这个魔改 https://github.com/rico93/ss-panel-v3-mod_Uim 面
> [image]

源码共享一下呗

WKZD, 2019-04-26:
> 我自己改修了，订阅增加支持v2ray和ss。 支持 这个魔改 https://github.com/rico93/ss-panel-v3-mod_Uim 面
> [image]
你改了的在哪？

closed 2019-09-29 by coolsnowwolf