coolsnowwolf/lede #1010: x86_64 编译报错

zcluo, 2019-01-15:
make[3]: Entering directory '/root/lede/feeds/packages/kernel/exfat-nofuse'
rm -f /root/lede/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5/.built
touch /root/lede/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5/.built_check
cat /dev/null > /root/lede/staging_dir/target-x86_64_musl/pkginfo/exfat-nofuse.symvers; for subdir in .; do cat /root/lede/staging_dir/target-x86_64_musl/pkginfo/*.symvers 2>/dev/null > /root/lede/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5/$subdir/Module.symvers; done
make -C "/root/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.9" ARCH="x86" CROSS_COMPILE="x86_64-openwrt-linux-musl-" M="/root/lede/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5" CONFIG_EXFAT_FS=m modules
make[4]: Entering directory '/root/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.9'
  CC [M]  /root/lede/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5/exfat_super.o
/root/lede/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5/exfat_super.c: In function 'exfat_fill_inode':
/root/lede/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5/exfat_super.c:1927:27: error: passing argument 2 of 'exfat_time_fat2unix' from incompatible pointer type [-Werror=incompatible-pointer-types]
  exfat_time_fat2unix(sbi, &inode->i_mtime, &info.ModifyTimestamp);
                           ^
/root/lede/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5/exfat_super.c:150:6: note: expected 'struct timespec *' but argument is of type 'struct timespec64 *'
 void exfat_time_fat2unix(struct exfat_sb_info *sbi, struct timespec *ts,
      ^~~~~~~~~~~~~~~~~~~
/root/lede/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5/exfat_super.c:1928:27: error: passing argument 2 of 'exfat_time_fat2unix' from incompatible pointer type [-Werror=incompatible-pointer-types]
  exfat_time_fat2unix(sbi, &inode->i_ctime, &info.CreateTimestamp);
                           ^
/root/lede/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5/exfat_super.c:150:6: note: expected 'struct timespec *' but argument is of type 'struct timespec64 *'
 void exfat_time_fat2unix(struct exfat_sb_info *sbi, struct timespec *ts,
      ^~~~~~~~~~~~~~~~~~~
/root/lede/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5/exfat_super.c:1929:27: error: passing argument 2 of 'exfat_time_fat2unix' from incompatible pointer type [-Werror=incompatible-pointer-types]
  exfat_time_fat2unix(sbi, &inode->i_atime, &info.AccessTimestamp);
                           ^
/root/lede/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5/exfat_super.c:150:6: note: expected 'struct timespec *' but argument is of type 'struct timespec64 *'
 void exfat_time_fat2unix(struct exfat_sb_info *sbi, struct timespec *ts,
      ^~~~~~~~~~~~~~~~~~~
/root/lede/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5/exfat_super.c: In function 'exfat_write_inode':
/root/lede/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5/exfat_super.c:2010:27: error: passing argument 2 of 'exfat_time_unix2fat' from incompatible pointer type [-Werror=incompatible-pointer-types]
  exfat_time_unix2fat(sbi, &inode->i_mtime, &info.ModifyTimestamp);
                           ^
/root/lede/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5/exfat_super.c:169:6: note: expected 'struct timespec *' but argument is of type 'struct timespec64 *'
 void exfat_time_unix2fat(struct exfat_sb_info *sbi, struct timespec *ts,
      ^~~~~~~~~~~~~~~~~~~
/root/lede/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5/exfat_super.c:2011:27: error: passing argument 2 of 'exfat_time_unix2fat' from incompatible pointer type [-Werror=incompatible-pointer-types]
  exfat_time_unix2fat(sbi, &inode->i_ctime, &info.CreateTimestamp);
                           ^
/root/lede/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5/exfat_super.c:169:6: note: expected 'struct timespec *' but argument is of type 'struct timespec64 *'
 void exfat_time_unix2fat(struct exfat_sb_info *sbi, struct timespec *ts,
      ^~~~~~~~~~~~~~~~~~~
/root/lede/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5/exfat_super.c:2012:27: error: passing argument 2 of 'exfat_time_unix2fat' from incompatible pointer type [-Werror=incompatible-pointer-types]
  exfat_time_unix2fat(sbi, &inode->i_atime, &info.AccessTimestamp);
                           ^
/root/lede/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5/exfat_super.c:169:6: note: expected 'struct timespec *' but argument is of type 'struct timespec64 *'
 void exfat_time_unix2fat(struct exfat_sb_info *sbi, struct timespec *ts,
      ^~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[5]: *** [scripts/Makefile.build:304: /root/lede/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5/exfat_super.o] Error 1
make[4]: *** [Makefile:1515: _module_/root/lede/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5] Error 2
make[4]: Leaving directory '/root/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.9'
make[3]: *** [Makefile:54: /root/lede/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5/.built] Error 2
make[3]: Leaving directory '/root/lede/feeds/packages/kernel/exfat-nofuse'
time: package/feeds/packages/exfat-nofuse/compile#1.42#0.31#1.64
make[2]: *** [package/Makefile:109: package/feeds/packages/exfat-nofuse/compile] Error 2
make[2]: Leaving directory '/root/lede'
make[1]: *** [package/Makefile:103: /root/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/root/lede'
make: *** [/root/lede/include/toplevel.mk:218: world] Error 2

opsiff, 2019-01-15:
compile error

ope087, 2019-01-21:
exfat驱动在4.18以上内核编译错误,openwrt的官方源有更新了
https://github.com/openwrt/packages/tree/master/kernel/exfat-nofuse