coolsnowwolf/lede #1012: k3使用中平均负载有些大。

huangjackey1111, 2019-01-15:
使用(git-18.296.11678-2bbc1e3) 编译的k3固件，之前使用luci-app-nlbwmon的时候，负载到了3.1+。之后卸载luci-app-nlbwmon，负载降到2.0+，但是还是很高的样子。不知道是什么地方出了问题。CPU使用率也不算高，排在前几的目前是softethervpn里面的程序，虽然也不高，但是总体负载还是挺高的。
编译的配置文件也上传了。
[image]
[image]

[config.txt](https://github.com/coolsnowwolf/lede/files/2759857/config.txt)

coolsnowwolf, 2019-01-18:
看负载的这个页面很占负载，一直看一直高

huangjackey1111, 2019-01-20:
> 看负载的这个页面很占负载，一直看一直高

但是直接用top命令来看，一进去就1.5-2.1了

huangjackey1111, 2019-01-20:
> 是不是有高io做程序在跑 zcluo
> […](#)

看io情况需要怎样查看？

hcym, 2019-01-20:
target.mk里面看看
除了ssr，v2ray其它的花花草草我是都去掉的

huangjackey1111, 2019-01-21:
> target.mk里面看看
> 除了ssr，v2ray其它的花花草草我是都去掉的

用大雕的固件就是因为可定制和装软件各种玩嘛~~
./include/target.mk
[Uploading include_target.txt…]()

huangjackey1111, 2019-01-21:
> target.mk里面看看
> 除了ssr，v2ray其它的花花草草我是都去掉的

用大雕的固件就是因为可定制和装软件各种玩嘛~~
./include/target.mk
[include_target.txt](https://github.com/coolsnowwolf/lede/files/2777260/include_target.txt)

closed 2019-09-29 by coolsnowwolf