coolsnowwolf/lede #1013: x86-64, 4.19内核,  failed to register a conntrack notifier

jackytang, 2019-01-16:
[    0.000000] Linux version 4.19.9 (vagrant@ubuntu-16) (gcc version 7.4.0 (OpenWrt GCC 7.4.0 r1164-730afc3)) #0 SMP Tue Jan 15 02:39:04 2019
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=3f57b8bd-02 rootfstype=squashfs rootwait console=tty0 console=ttyS0,115200n8 noinitrd
[    0.000000] x86/fpu: x87 FPU will use FXSAVE
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009c7ff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009c800-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000001effffff] usable
[    0.000000] BIOS-e820: [mem 0x000000001f000000-0x00000000201fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000020200000-0x000000007a8e0fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007a8e1000-0x000000007b0d8fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007b0d9000-0x000000007b0e8fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007b0e9000-0x000000007b671fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007b672000-0x000000007b991fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007b992000-0x000000007bffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fea00000-0x00000000feafffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed03000-0x00000000fed03fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed06000-0x00000000fed06fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed08000-0x00000000fed09fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1cfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed80000-0x00000000fedbffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffb00000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000017fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 3.0.0 present.
[    0.000000] DMI: Default string Default string/Aptio CRB, BIOS 5.11 06/23/2016
[    0.000000] tsc: Detected 1600.000 MHz processor
[    0.003179] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.003187] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.003206] last_pfn = 0x180000 max_arch_pfn = 0x400000000
[    0.003214] MTRR default type: uncachable
[    0.003215] MTRR fixed ranges enabled:
[    0.003218]   00000-9FFFF write-back
[    0.003220]   A0000-BFFFF uncachable
[    0.003223]   C0000-FFFFF write-protect
[    0.003226] MTRR variable ranges enabled:
[    0.003229]   0 base 000000000 mask F80000000 write-back
[    0.003231]   1 base 07C000000 mask FFC000000 uncachable
[    0.003234]   2 base 100000000 mask F80000000 write-back
[    0.003236]   3 base 090000000 mask FF0000000 write-through
[    0.003238]   4 base 0A0000000 mask FFFC00000 write-through
[    0.003240]   5 disabled
[    0.003241]   6 disabled
[    0.003242]   7 disabled
[    0.003535] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.003821] last_pfn = 0x7c000 max_arch_pfn = 0x400000000
[    0.011807] found SMP MP-table at [mem 0x000fcc60-0x000fcc6f] mapped at [(____ptrval____)]
[    0.011834] Base memory trampoline at [(____ptrval____)] 96000 size 24576
[    0.011846] BRK [0x02201000, 0x02201fff] PGTABLE
[    0.011853] BRK [0x02202000, 0x02202fff] PGTABLE
[    0.011856] BRK [0x02203000, 0x02203fff] PGTABLE
[    0.012070] BRK [0x02204000, 0x02204fff] PGTABLE
[    0.012765] BRK [0x02205000, 0x02205fff] PGTABLE
[    0.012951] BRK [0x02206000, 0x02206fff] PGTABLE
[    0.013765] ACPI: Early table checksum verification disabled
[    0.013965] ACPI: RSDP 0x00000000000F05B0 000024 (v02 DELL  )
[    0.013976] ACPI: XSDT 0x000000007B0E0088 00008C (v01 DELL   PE_SC3   01072009 AMI  00010013)
[    0.013990] ACPI: FACP 0x000000007B0E7928 00010C (v05 DELL   PE_SC3   01072009 AMI  00010013)
[    0.014003] ACPI BIOS Warning (bug): 32/64X length mismatch in FADT/Gpe0Block: 128/32 (20180810/tbfadt-569)
[    0.014016] ACPI: DSDT 0x000000007B0E01A0 007782 (v02 DELL   PE_SC3   01072009 INTL 20120913)
[    0.014026] ACPI: FACS 0x000000007B659E80 000040
[    0.014034] ACPI: APIC 0x000000007B0E7A38 000084 (v03 DELL   PE_SC3   01072009 AMI  00010013)
[    0.014043] ACPI: FPDT 0x000000007B0E7AC0 000044 (v01 DELL   PE_SC3   01072009 AMI  00010013)
[    0.014052] ACPI: FIDT 0x000000007B0E7B08 00009C (v01 DELL   PE_SC3   01072009 AMI  00010013)
[    0.014062] ACPI: MCFG 0x000000007B0E7BA8 00003C (v01 DELL   PE_SC3   01072009 MSFT 00000097)
[    0.014071] ACPI: SLIC 0x000000007B0E7BE8 000176 (v01 DELL   PE_SC3   01072009 AMI  00010013)
[... 541 lines cut ...]
[   32.043815] br-lan: port 3(eth3) entered blocking state
[   32.049889] br-lan: port 3(eth3) entered forwarding state
[   32.058562] RTL8211E Gigabit Ethernet r8169-100:00: attached PHY driver [RTL8211E Gigabit Ethernet] (mii_bus:phy_addr=r8169-100:00, irq=IGNORE)
[   32.319846] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   32.320914] r8169 0000:01:00.0 eth0: Link is Down
[   32.347576] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[   32.517972] br-lan: port 2(eth2) entered disabled state
[   32.524213] br-lan: port 3(eth3) entered disabled state
[   33.661318] r8169 0000:02:00.0 eth1: Link is Up - 1Gbps/Full - flow control rx/tx
[   33.670272] br-lan: port 1(eth1) entered blocking state
[   33.676352] br-lan: port 1(eth1) entered forwarding state
[   34.256009] r8169 0000:04:00.0 eth3: Link is Up - 1Gbps/Full - flow control rx/tx
[   34.264947] br-lan: port 3(eth3) entered blocking state
[   34.271130] br-lan: port 3(eth3) entered forwarding state
[   34.351946] r8169 0000:03:00.0 eth2: Link is Up - 1Gbps/Full - flow control rx/tx
[   34.360890] br-lan: port 2(eth2) entered blocking state
[   34.366969] br-lan: port 2(eth2) entered forwarding state
[   36.234997] r8169 0000:01:00.0 eth0: Link is Up - 1Gbps/Full - flow control off
[   36.243739] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   36.335197] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[   36.599838] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[   41.466215] pppoe-wan: renamed from ppp0
[   44.598354] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[   44.973487] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
[   74.565338] random: crng init done

zcluo, 2019-01-16:
我的也是这个问题 请问解决了吗

dongjuanyong, 2019-01-18:
4.19的内核少加了一个patch，把
lede/target/linux/generic/hack-4.14/952-net-conntrack-events-support-multiple-registrant.patch
这个4.14的patch，cp到
lede/target/linux/generic/hack-4.19/
重新编译就行了，已测试x86_64的4.19一切正常，这个patch虽然是博通给SFE设计的，但是fullconenat也用得上