coolsnowwolf/lede #1019: v2ray 缺少 socks/http 代理，我做了个补丁可以用

qiang-yu, 2019-01-18:
目前固件中 v2ray 只支持 透明代理，不支持单独开启一个 socks/http 代理

有时候我们需要的网站不在 gfwlist 里面，我们需要临时访问一下，最简单的方法就是 v2ray 有一个 socks/http 代理， 我们临时代理过去就行

我修改了代码，v2ray 自动开启 socks/http 代理， 你设置转发端口是 port ，那么 socks 代理端口就是 port +1 ，  http 代理端口就是  port +2

比如你在  ShadowsocksR Plus+ 里面设置了 v2ray 的端口是 1079 (透明转发端口)，那么对应的 socks 代理端口就是 1080， http 代理端口就是  1081

使用方法：

把附件中的文件下载，解压缩，得到一个   genv2config.lua  文件，把这个文件覆盖到代码目录 package/lean/luci-app-ssr-plus/root/usr/share/shadowsocksr/genv2config.lua  然后重新打包编译就可以了

或者，不编译的话，也可以直接覆盖到固件 /usr/share/shadowsocksr/genv2config.lua 上直接使用

希望 L大 后面能加上 v2ray 设置 socks/http 代理的功能，就不用像我这样打补丁去用了

[genv2config.zip](https://github.com/coolsnowwolf/lede/files/2773519/genv2config.zip)

forsan-co, 2019-03-11:
这个可以，感觉可以提交下代码给大屌

closed 2019-10-18 by coolsnowwolf

klever1988, 2019-11-06:
这个好用，大雕帮忙merge了吧

h6fYe34KFSiH, 2019-12-03:
还有个曲线救国的方法，就是你在 "ShadowsocksR 服务端" 新建一个用户，然后去客户端添加这个ssr节点，然后在”高级设置开启socks5“代理，选择这个ssr节点。

相当于使用路由里的ssr服务端中转了v2ray，ssr是可以开启socks5代理的。
又可以开心地使用switchyomega了！

h6fYe34KFSiH, 2019-12-05:
我替换之后 ssrplus 无法运行了。。