coolsnowwolf/lede #1036: 编译到noddos报错

molichen88, 2019-01-30:
make[3]: Entering directory `/home/chen/lede/feeds/packages/net/noddos'
touch /home/chen/lede/build_dir/target-x86_64_musl/noddos-0.5.5/.prepared_4dc22ddbaf7be18bd26f0e3b5f84dcf4_6664517399ebbbc92a37c5bb081b5c53_check
. /home/chen/lede/include/shell.sh; xzcat /home/chen/lede/dl/noddos-0.5.5.tar.xz | tar -C /home/chen/lede/build_dir/target-x86_64_musl/noddos-0.5.5/.. -xf - 
[ ! -d ./src/ ] || cp -fpR ./src/. /home/chen/lede/build_dir/target-x86_64_musl/noddos-0.5.5
touch /home/chen/lede/build_dir/target-x86_64_musl/noddos-0.5.5/.prepared_4dc22ddbaf7be18bd26f0e3b5f84dcf4_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/chen/lede/build_dir/target-x86_64_musl/noddos-0.5.5/.configured_*
rm -f /home/chen/lede/staging_dir/target-x86_64_musl/stamp/.noddos_installed
mkdir -p /home/chen/lede/build_dir/target-x86_64_musl/noddos-0.5.5
(cd /home/chen/lede/build_dir/target-x86_64_musl/noddos-0.5.5; CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/chen/lede/build_dir/target-x86_64_musl/noddos-0.5.5:noddos-0.5.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/chen/lede/build_dir/target-x86_64_musl/noddos-0.5.5:noddos-0.5.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " LDFLAGS="-L/home/chen/lede/staging_dir/target-x86_64_musl/usr/lib -L/home/chen/lede/staging_dir/target-x86_64_musl/lib -L/home/chen/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/usr/lib -L/home/chen/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/lib -znow -zrelro " cmake -DCMAKE_SYSTEM_NAME=Linux -DCMAKE_SYSTEM_VERSION=1 -DCMAKE_SYSTEM_PROCESSOR=x86_64 -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_CXX_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_C_COMPILER="/home/chen/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/bin/x86_64-openwrt-linux-musl-gcc" -DCMAKE_C_COMPILER_ARG1="" -DCMAKE_CXX_COMPILER="/home/chen/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/bin/x86_64-openwrt-linux-musl-g++" -DCMAKE_CXX_COMPILER_ARG1="" -DCMAKE_ASM_COMPILER="/home/chen/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/bin/x86_64-openwrt-linux-musl-gcc" -DCMAKE_ASM_COMPILER_ARG1="" -DCMAKE_EXE_LINKER_FLAGS:STRING="-L/home/chen/lede/staging_dir/target-x86_64_musl/usr/lib -L/home/chen/lede/staging_dir/target-x86_64_musl/lib -L/home/chen/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/usr/lib -L/home/chen/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/lib -znow -zrelro" -DCMAKE_MODULE_LINKER_FLAGS:STRING="-L/home/chen/lede/staging_dir/target-x86_64_musl/usr/lib -L/home/chen/lede/staging_dir/target-x86_64_musl/lib -L/home/chen/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/usr/lib -L/home/chen/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/lib -znow -zrelro -Wl,-Bsymbolic-functions" -DCMAKE_SHARED_LINKER_FLAGS:STRING="-L/home/chen/lede/staging_dir/target-x86_64_musl/usr/lib -L/home/chen/lede/staging_dir/target-x86_64_musl/lib -L/home/chen/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/usr/lib -L/home/chen/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/lib -znow -zrelro -Wl,-Bsymbolic-functions" -DCMAKE_AR="/home/chen/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/bin/x86_64-openwrt-linux-musl-gcc-ar" -DCMAKE_NM="/home/chen/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/bin/x86_64-openwrt-linux-musl-gcc-nm" -DCMAKE_RANLIB="/home/chen/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/bin/x86_64-openwrt-linux-musl-gcc-ranlib" -DCMAKE_FIND_ROOT_PATH="/home/chen/lede/staging_dir/target-x86_64_musl/usr;/home/chen/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl" -DCMAKE_FIND_ROOT_PATH_MODE_PROGRAM=BOTH -DCMAKE_FIND_ROOT_PATH_MODE_LIBRARY=ONLY -DCMAKE_FIND_ROOT_PATH_MODE_INCLUDE=ONLY -DCMAKE_STRIP=: -DCMAKE_INSTALL_PREFIX=/usr -DDL_LIBRARY=/home/chen/lede/staging_dir/target-x86_64_musl -DCMAKE_PREFIX_PATH=/home/chen/lede/staging_dir/target-x86_64_musl -DCMAKE_SKIP_RPATH=TRUE  /home/chen/lede/build_dir/target-x86_64_musl/noddos-0.5.5 )
-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is GNU 7.4.0
-- Check for working C compiler: /home/chen/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/bin/x86_64-openwrt-linux-musl-gcc
-- Check for working C compiler: /home/chen/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/bin/x86_64-openwrt-linux-musl-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/chen/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/bin/x86_64-openwrt-linux-musl-g++
-- Check for working CXX compiler: /home/chen/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/bin/x86_64-openwrt-linux-musl-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    CMAKE_ASM_COMPILER
    CMAKE_ASM_COMPILER_ARG1
    CMAKE_FIND_ROOT_PATH_MODE_LIBRARY
    CMAKE_MODULE_LINKER_FLAGS
    CMAKE_SHARED_LINKER_FLAGS
    DL_LIBRARY

-- Build files have been written to: /home/chen/lede/build_dir/target-x86_64_musl/noddos-0.5.5
touch /home/chen/lede/build_dir/target-x86_64_musl/noddos-0.5.5/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/chen/lede/build_dir/target-x86_64_musl/noddos-0.5.5/.built
touch /home/chen/lede/build_dir/target-x86_64_musl/noddos-0.5.5/.built_check
CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/chen/lede/build_dir/target-x86_64_musl/noddos-0.5.5:noddos-0.5.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/chen/lede/staging_dir/target-x86_64_musl/usr/include -I/home/chen/lede/staging_dir/target-x86_64_musl/include -I/home/chen/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/usr/include -I/home/chen/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/include/fortify -I/home/chen/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/include " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/chen/lede/build_dir/target-x86_64_musl/noddos-0.5.5:noddos-0.5.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/chen/lede/staging_dir/target-x86_64_musl/usr/include -I/home/chen/lede/staging_dir/target-x86_64_musl/include -I/home/chen/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/usr/include -I/home/chen/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/include/fortify -I/home/chen/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/include " LDFLAGS="-L/home/chen/lede/staging_dir/target-x86_64_musl/usr/lib -L/home/chen/lede/staging_dir/target-x86_64_musl/lib -L/home/chen/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/usr/lib -L/home/chen/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/lib -znow -zrelro " make -j1 -C /home/chen/lede/build_dir/target-x86_64_musl/noddos-0.5.5/. AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/chen/lede/build_dir/target-x86_64_musl/noddos-0.5.5:noddos-0.5.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CROSS="x86_64-openwrt-linux-musl-" ARCH="x86_64" CMAKE_COMMAND='/home/chen/lede/staging_dir/host/bin/cmake' CMAKE_DISABLE_cmake_check_build_system=1 ;
make[4]: Entering directory `/home/chen/lede/build_dir/target-x86_64_musl/noddos-0.5.5'
make[5]: Entering directory `/home/chen/lede/build_dir/target-x86_64_musl/noddos-0.5.5'
make[6]: Entering directory `/home/chen/lede/build_dir/target-x86_64_musl/noddos-0.5.5'
[... cut ...]
make[4]: Leaving directory `/home/chen/lede/build_dir/target-x86_64_musl/noddos-0.5.5'
make[3]: *** [/home/chen/lede/build_dir/target-x86_64_musl/noddos-0.5.5/.built] Error 2
make[3]: Leaving directory `/home/chen/lede/feeds/packages/net/noddos'
time: package/feeds/packages/noddos/compile#6.51#0.77#8.93
make[2]: *** [package/feeds/packages/noddos/compile] Error 2
make[2]: Leaving directory `/home/chen/lede'
make[1]: *** [/home/chen/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/chen/lede'
make: *** [world] Error 2

blueveryday, 2019-02-13:
我也遇到了，还以为只有我有问题。
noddos怎么编译都不通过。

blueveryday, 2019-02-14:
试过了，还是不行。
我把lede文件夹删除，重新clone，重新编码。
只要选择noddos，就会编译报错。
而且取消以后，编译还是会报这个错，晕死了。

hzjnet, 2019-04-24:
2019年版本确实存在noddos和wifidog-ng编译问题

closed 2019-09-29 by coolsnowwolf