coolsnowwolf/lede #1057: OpenWrt 18.06.2 -> No package 'protobuf' found

p3x-robot, 2019-02-11:
Hello!
How are you?
It is still not working, although they are saying it is fixed.
But for me it looks like I cannot create the firmwares (multple targets) and probably is because I cannot build protobuf and profobuf-c, the error as it was:
```text
make[3]: Entering directory '/build/source/feeds/packages/libs/protobuf-c'

.........
etc......
.........

checking for protobuf >= 2.6.0... no
configure: error: Package requirements (protobuf >= 2.6.0) were not met:

No package 'protobuf' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables protobuf_CFLAGS
and protobuf_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
Makefile:67: recipe for target '/build/source/build_dir/hostpkg/libprotobuf-c-v1.2.1/.configured' failed
make[3]: *** [/build/source/build_dir/hostpkg/libprotobuf-c-v1.2.1/.configured] Error 1
make[3]: Leaving directory '/build/source/feeds/packages/libs/protobuf-c'
time: package/feeds/packages/protobuf-c/host-compile#10.72#0.90#58.90
```

How can I make it to make it work?

https://forum.openwrt.org/t/collected-errors-satisfy-dependencies-for/31077/10

Based on the root error is:
```text
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-cfg80211:
 *      iw
 * opkg_install_cmd: Cannot install package kmod-cfg80211.
```

closed 2019-02-16 by p3x-robot

rareshornet, 2019-06-10:
@p3x-robot did you found a way to make it work? This is still an issue apparently :/

p3x-robot, 2019-06-10:
i could not solve this.

SinglWolf, 2019-07-02:
> i could not solve this.

I solved this problem like this. I Launched in the console compilation of one protobuf package.
`make package/protobuf/{clean,compile} V=s -j4`
Then I continued to compile the firmware.
`make -j4`
Everything went without error.