coolsnowwolf/lede #1085: 为什么PS4下载强制走代理？

forever8938, 2019-02-25:
PS4下载服务器没有被墙，gfwlist里不含任何playstation的网址。ssrplus选的是gfw模式。看了下代码，似乎ssrplus自带的gfwlist里有playstation的网址。每天定时更新gfwlist没覆盖掉默认的规则吗，只是合并而已？

sqliuchang, 2019-02-26:
是合并的啊，要不然自己添加的gfwlist岂不是被覆盖了

forever8938, 2019-02-26:
每天自动更新的gfwlist要覆盖默认规则，然后再和自己添加的合并。默认规则不覆盖，岂不是里面的网址都强制代理？我就问gfw模式下载PS4如何不走代理？这很明显的问题，PSN又没被墙，我也没添加规则。

forever8938, 2019-02-26:
建议删除gfw_base.conf，启动后下载gfwlist，保存到闪存。gfwlist不可能下载不了的，实在不行走代理，没代理还翻啥墙。
或者把gfw_base.conf更新一下。

SouHong, 2019-02-26:
建议使用ssr-gfw最新源码更新，另外删除gfw_base.conf文件或者把它修改成更新后的gfw_list.conf列表。我是删除了gfw_base.conf并且删除了/etc/config/gfw.list文件中与gfw_list.conf列表重复的地址。现在gfw_list.conf列表5165条，/etc/config/gfw.list列表33条。

coolsnowwolf, 2019-03-01:
具体是哪些网址知道吗？

forever8938, 2019-03-01:
L大你好。问题就是实机上的/etc/dnsmasq.ssr/gfw_base.conf内容太旧，只要根据最新gfwlist生成这个文件即可解决问题。我不清楚代码，搜了github没找到gfw_base.conf。

forever8938, 2019-03-02:
gfw_list.conf是一开始就有的，每天还会自动更新。
gfw_base.conf万年不变，还会合并到规则里，所以完全没必要。

为了避免出问题，个人建议这样改：
1. 清空gfw_base.conf，保留文件。代码逻辑不变。
2. 更新gfw_list.conf

供参考。

closed 2019-04-17 by coolsnowwolf