coolsnowwolf/lede #1094: SSR plus+ 开启ss透明代理之后，对于ipv6还是无效

jindaxia, 2019-03-04:
附测试IPV6地址
https://test-ipv6.com/ 

环境：
本地只有IPV4环境，通过ssr plus增加ss透明代理之后，采用gfw模式
默认打开上面测试网址，因为没在gfw列表中，不代理，所以不支持ipv6也正常，
设置全局代理LAN IP 为本机之后  打开测试网站，IPV4地址显示的是我的代理IP，但是ipv6不正常

本地不通过路由器代理，直接用ss客户端连接，访问测试网站 可以支持完整的ipv6功能

dongjuanyong, 2019-03-04:
第一个问题，直接用ss客户端连接，访问测试网站 可以支持完整的ipv6功能，这时是开的全局么？
第二个问题，同样是直接用ss客户端连接，访问测试网站，这是上面测试网站上显示的ISP是你代理服务器的ISP还是你本地的？
第三个问题，你是否确定你的代理服务器支持IPV6协议？
#990 我简单了解过，要实现ipv6透明代理，需要和ipv4的udp一样通过tproxy，理论上可行，但会出现icmp不通的问题，这某种程度上对ipv6是致命的，再加上我服务器没有v6环境，所以无法尝试
注意：socks/http代理和透明代理实现起来差别很大，所以并不是客户端能干的事透明代理都能很简单的解决

jindaxia, 2019-03-05:
1. 开了全局
2. 显示的是我代理服务器IP
3. 我的服务器支持IPV6，有ipv6地址

dongjuanyong, 2019-03-05:
还是那句话，你搞不明白socks/http代理和透明代理实现差别的话，
使用ss客户端的时候，因为本地请求是指向客户端的监听端口的（通过软件更改系统的设置），通过socks/http打包和ss加密发送至代理服务器，对测试网站来说相当于一切请求都来自代理服务器。而你的代理服务器支持v6协议，所以各方面正常，本质上是在测你的代理服务器和测试网站之间的v6连通性
使用透明代理的时候，网关（路由器）识别截获来自客户端的流量（请求和流量之间是有差别的），重定向至运行在网关上的ss客户端的监听端口，后边基本一样。识别截获流量依靠的是iptables上的规则，目前据我所知，所有的规则只适用于v4（tcp+udp），没有人给出过完整可以正常运行的重定向v6协议的iptables规则，所以只能自己丰衣足食。再加上你本地没有v6环境，客户端本身不会有任何v6的流量发出，网关上想做什么都做不到

jindaxia, 2019-03-05:
OK，我理解，现在本地电脑和路由器之间是有v6地址的
问题：
是不是需要在路由器上通过配置 ip6tables规则，转发v6的流量到 ss-redir的1234端口上？而ss-redir的监听地址需要改成 `::0`,我会继续做一些尝试的

dongjuanyong, 2019-03-05:
思路没问题，监听设置应该是["[::0]","0.0.0.0"]这样的，只有[::0]的话v4不通

dongjuanyong, 2019-03-05:
另外我有个坏消息
https://github.com/shadowsocks/shadowsocks-libev/issues/1982
ss-redir只支持udp走tproxy，这就意味着你必须靠nat6才能做tcp v6的透明代理，nat6这个事总觉得与ipv6的初衷向违背
似乎v2ray的tproxy没这个限制，不过我是没机会试了

closed 2019-09-29 by coolsnowwolf

573329646, 2020-12-08:
> 附测试IPV6地址
> https://test-ipv6.com/
> 
> 环境：
> 本地只有IPV4环境，通过ssr plus增加ss透明代理之后，采用gfw模式
> 默认打开上面测试网址，因为没在gfw列表中，不代理，所以不支持ipv6也正常，
> 设置全局代理LAN IP 为本机之后 打开测试网站，IPV4地址显示的是我的代理IP，但是ipv6不正常
> 
> 本地不通过路由器代理，直接用ss客户端连接，访问测试网站 可以支持完整的ipv6功能，说明服务端设置没问题

楼主有解决吗，我也遇到了这样的问题