coolsnowwolf/lede #1101: 有人已经搞定k3屏幕天气了，lean来一发？

shenzt68, 2019-03-07:
基于updateing的代码修改，改动很少，主要是弄清楚数据结构费点时间，从官方的16进制log分析出来的，有需要的自己根据数据结构改就行了

typedef struct _weather_info {
    char city[32]; //成都
    char temp[4]; //温度11
    char date[11]; //日期2019-02-20
    char time[6]; //时间14:29
    unsigned char weather; //weather_code
    unsigned char week; //0 1 2 3 4 5 6
    unsigned char error; //0
} WEATHER_INFO;
https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=466672&highlight=%CC%EC%C6%F8

coolsnowwolf, 2019-03-12:
来一发

lin20121221, 2019-03-21:
https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=502489&extra=page%3D1%26filter%3Dtypeid%26typeid%3D15

zxlhhyccc, 2019-04-19:
> https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=502489&extra=page%3D1%26filter%3Dtypeid%26typeid%3D15

我已经搞定了屏幕了，每个屏幕均正常显示。。。。

closed 2019-09-29 by coolsnowwolf