coolsnowwolf/lede #1106: L大，可以给clash写一个luci吗？

Anyfc, 2019-03-08:
项目地址：https://github.com/Dreamacro/clash

Clash：一个 GO 开发的、基于规则的多平台代理客户端，兼容 Shadowsocks、V2Ray 等协议，拥有像 Surge 一样强大的代理规则
使用 HTTP/HTTPS and SOCKS
和 Surge 相似的配置
支持基于地域的规则
支持 Vmess/Shadowsocks/Socks5 服务端协议
支持基于 Netfilter TCP 流量重定向

近乎完美的clash，L大，可以给clash写一个luci配合使用吗？

caonimac, 2019-03-09:
你的这个不就是ssr plus吗？楼主这是看不起ssr plus吗？哈哈哈。

Anyfc, 2019-03-09:
并不是，ssr plus我也在用，ssr plus不能同时运行Vmess/Shadowsocks/Socks5 服务端协议，只能单独运行某一个服务端协议，而clash可以，不知道你有没有用过iOS端的代理app，clash是可以像iOS端的app一样设置代理规则的，比如：
aaa.com 设置走 shadowsocks 的代理
bbb.com 设置走Vmess的代理
ccc.com 设置走HTTP/HTTPS的代理

通俗点说就是，我可以设置策略：
 YouTube 走流量多一个shadowsocks1
twitter 走速度快的Vmess
Facebook 走另一个shadowsocks2

coolsnowwolf, 2019-03-11:
路由端因为要适配 8M SPI ROM + 64M RAM 的路由器，所以无法做这样的一个客户端。

技术上来说并不存在太大的问题。

Anyfc, 2019-03-11:
@coolsnowwolf 可以做成ipk的形式提供给内存和空间大的路由器的用户吗？像v2ray哪样的，v2ray的ipk有近10M大小，而clash使用upx压缩一下不到3M，加上Country.mmdb，总共不到6M

coolsnowwolf, 2019-03-11:
8M ROM 还剩下 100KB，不能超过这个

arfaWong, 2019-03-25:
https://github.com/505575853/lede/tree/R9/package/my/luci-app-clash

505575853, 2019-04-06:
@arfaWong 不要发我的那个，启动脚本错误

hcym, 2019-04-06:
n1上路，体积不是问题，看各位大侠侣出手

loveqianool, 2019-04-06:
clash 配置中有外部控制功能
allow-lan: true
external-controller: '127.0.0.1:9090' 
然后使用 http://clash.razord.top 或 http://yacd.haishan.me 进行配置.

vernesong, 2019-04-28:
https://github.com/frainzy1477/clash
这个项目编译后替换二进制文件在wndr3800上可用，不装太多插件的话空间没问题，目前项目自带的web管理我这无法登陆，可能是因为依赖的关系。但使用上面的外部面板可以管理没问题。 @coolsnowwolf

Anyfc, 2019-05-05:
@vernesong 自带的web管理应该是没有放进去。。

vernesong, 2019-05-05:
> @vernesong 自带的web管理应该是没有放进去。。

不是这个原因，是面板不支持https的原因

wangxin6, 2019-05-16:
只要把calsh编译进固件就可以。配置可以在电脑端使用外部控制功能。

closed 2019-09-29 by coolsnowwolf

oldhalf, 2020-02-10:
好像有人开发了
https://github.com/frainzy1477/luci-app-clash