coolsnowwolf/lede #1111: X86-64编译出错

engineerlzk, 2019-03-13:
编译环境：windows10中的WSL子系统，安装的是ubuntu16.04，已经对lede目录设置了区分大小写，在正式编译前的各个环节（包括make menuconfig）都正常，辛辛苦苦编译了近6个小时，出现如下错误信息：
quilt.1.in -> quilt.1
/usr/bin/pod2man bin/guards > bin/guards.1
make[4]: *** No rule to make target 'compat/column.in', needed by 'po/quilt.pot'.  Stop.
make[4]: Leaving directory '/mnt/i/lede/build_dir/host/quilt-0.65'
Makefile:36: recipe for target '/mnt/i/lede/build_dir/host/quilt-0.65/.built' failed
make[3]: *** [/mnt/i/lede/build_dir/host/quilt-0.65/.built] Error 2
make[3]: Leaving directory '/mnt/i/lede/tools/quilt'
time: tools/quilt/compile#2.01#37.17#103.90
tools/Makefile:155: recipe for target 'tools/quilt/compile' failed
make[2]: *** [tools/quilt/compile] Error 2
make[2]: Leaving directory '/mnt/i/lede'
tools/Makefile:153: recipe for target '/mnt/i/lede/staging_dir/target-x86_64_musl/stamp/.tools_compile_yynyyyyynyyyyynyynnnyyyynyyyyyyyyyyyyyyynyyynynnyyynnyy' failed
make[1]: *** [/mnt/i/lede/staging_dir/target-x86_64_musl/stamp/.tools_compile_yynyyyyynyyyyynyynnnyyyynyyyyyyyyyyyyyyynyyynynnyyynnyy] Error 2
make[1]: Leaving directory '/mnt/i/lede'
/mnt/i/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

请问是哪个地方出了问题，如何破？谢谢指教！

gavin219, 2019-03-13:
权限问题。home目录应该就行了

engineerlzk, 2019-03-13:
意思是把lede下载到home目录再编译？谢谢！

SasamiMasakiJurai, 2019-03-15:
recipe for target 'tools/quilt/compile' failed

这个应该是tools下的quilt出错

找找看dl下是否有quilt在

有的话删除掉

让系统重新下载

最好在编译前用

make download -j1 V=99

engineerlzk, 2019-03-16:
> recipe for target 'tools/quilt/compile' failed
> 
> 这个应该是tools下的quilt出错
> 
> 找找看dl下是否有quilt在
> 
> 有的话删除掉
> 
> 让系统重新下载
> 
> 最好在编译前用
> 
> make download -j1 V=99

谢谢，我去试一下

engineerlzk, 2019-03-19:
最后的解决方案：翻看编译错误日志中出错的是哪些工具包，单独把这些文件事先重新下载下来，然后拷贝到/mnt/i/lede/dl目录，再用make编译就成功了！

closed 2019-03-22 by coolsnowwolf