coolsnowwolf/lede #1116: Newifi D2编译出错

1715173329, 2019-03-16:
所有的操作都是按照步骤一步步来的，最后出现这个错误。。
```
/bin/bash: line 1:  3689 Killed                  build/genautomata /home/router/lede/build_dir/toolchain-mipsel_24kc_gcc-7.4.0_musl/gcc-7.4.0/gcc/common.md /home/routerc_gcc-7.4.0_musl/gcc-7.4.0/gcc/config/mips/mips.md insn-conditions.md > tmp-automata.c
make[5]: *** [s-automata] Error 137
make[5]: Leaving directory `/home/router/lede/build_dir/toolchain-mipsel_24kc_gcc-7.4.0_musl/gcc-7.4.0-initial/gcc'
make[4]: *** [all-gcc] Error 2
make[4]: Leaving directory `/home/router/lede/build_dir/toolchain-mipsel_24kc_gcc-7.4.0_musl/gcc-7.4.0-initial'
make[3]: *** [/home/router/lede/build_dir/toolchain-mipsel_24kc_gcc-7.4.0_musl/gcc-7.4.0-initial/.built] Error 2
make[3]: Leaving directory `/home/router/lede/toolchain/gcc/initial'
time: toolchain/gcc/initial/compile#192.82#51.17#257.49
make[2]: *** [toolchain/gcc/initial/compile] Error 2
make[2]: Leaving directory `/home/router/lede'
make[1]: *** [/home/router/lede/staging_dir/toolchain-mipsel_24kc_gcc-7.4.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory `/home/router/lede'
make: *** [world] Error 2
```

ShadowTeamCN, 2019-03-17:
用vps编译?硬件配置不够吧，可能是OOM了，进程被kernel干掉了

1715173329, 2019-03-23:
> 用vps编译?硬件配置不够吧，可能是OOM了，进程被kernel干掉了

emm我换个其他的配置试试，谢谢你。

closed 2019-03-23 by 1715173329