coolsnowwolf/lede #1117: 有一个编译错误解决了好久，大神们帮忙看看

a1768623020, 2019-03-16:
. /home/liuhaoming/lede/include/shell.sh; xzcat /home/liuhaoming/lede/dl/automake-1.15.1.tar.xz | tar -C /home/liuhaoming/lede/build_dir/host/automake-1.15.1/.. -xf - 
tar: error while loading shared libraries: libiconv.so.2: cannot open shared object file: No such file or directory
Makefile:51: recipe for target '/home/liuhaoming/lede/build_dir/host/automake-1.15.1/.preparedeab14e1c26a942372a776ac004c558e4_6664517399ebbbc92a37c5bb081b5c53' failed
make[3]: *** [/home/liuhaoming/lede/build_dir/host/automake-1.15.1/.preparedeab14e1c26a942372a776ac004c558e4_6664517399ebbbc92a37c5bb081b5c53] Error 127
make[3]: Leaving directory '/home/liuhaoming/lede/tools/automake'
time: tools/automake/compile#0.05#0.06#45.82
tools/Makefile:155: recipe for target 'tools/automake/compile' failed
make[2]: *** [tools/automake/compile] Error 2
make[2]: Leaving directory '/home/liuhaoming/lede'
tools/Makefile:153: recipe for target '/home/liuhaoming/lede/staging_dir/target-mipsel_24kc_musl/stamp/.tools_compile_yynyyyyynyyyyynyynnyyyyynyyyyyyyyyyyyyyynnyynynnyyynnyy' failed
make[1]: *** [/home/liuhaoming/lede/staging_dir/target-mipsel_24kc_musl/stamp/.tools_compile_yynyyyyynyyyyynyynnyyyyynyyyyyyyyyyyyyyynnyynynnyyynnyy] Error 2
make[1]: Leaving directory '/home/liuhaoming/lede'
/home/liuhaoming/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

a1768623020, 2019-03-16:
> 目标'tools / automake / compile'的配方失败了
> 
> automake的出错
> 
> 查看DL下是否有automake的
> 
> 有的话删除
> 
> 让系统重新下载

这个文件删除下好几回了都是这个错误

SasamiMasakiJurai, 2019-03-16:
系统自己下载的？？？

是否是权限不够？？？

a1768623020, 2019-03-16:
> 系统自己下载的？？？
> 
> 是否是权限不够？？？

直接在home目录下Git的啊。。。能是权限不够吗？

SasamiMasakiJurai, 2019-03-16:
有概率的

我碰到过文件权限不够报错的。。。。。

可以的话

chmod -R 777 dl

给dl文件夹提升权限

a1768623020, 2019-03-16:
> 有概率的
> 
> 我碰到过文件权限不够报错的。。。。。
> 
> 可以的话
> 
> chmod -R 777 dl
> 
> 给dl文件夹提升权限

好像根本不是权限的问题。。

rampageX, 2019-03-16:
> tar: error while loading shared libraries: libiconv.so.2: cannot open shared object file: No such file or directory

第一部解压就失败了呀，试试用 `tar` 手动解个包看看。

a1768623020, 2019-03-16:
> > tar: error while loading shared libraries: libiconv.so.2: cannot open shared object file: No such file or directory
> 
> 第一步解压就失败了呀，试试用 `tar` 手动解个包看看。

tar解压没问题啊。。。能正常解包

qqj147258, 2019-04-22:
root用户了吧

hanxi, 2019-05-29:
系统是 Ubuntu 14.04 吗？

mengbl, 2019-09-04:
问题解决了么，我也是同样问题

closed 2019-09-29 by coolsnowwolf