coolsnowwolf/lede #1124: 一个文件无法下载（域名没有解析）导致编译失败

Creedowl, 2019-03-21:
http://www.mr511.de/software/libelf-0.8.13.tar.gz
无法解析
```
SHELL= flock /home/creedowl/lede/tmp/.libelf-0.8.13.tar.gz.flock -c '  	/home/creedowl/lede/scripts/download.pl "/home/creedowl/lede/dl" "libelf-0.8.13.tar.gz" "591a9b4ec81c1f2042a97aa60564e0cb79d041c52faa7416acb38bc95bd2c76d" "" "http://www.mr511.de/software/"    '
+ curl -f --connect-timeout 20 --retry 5 --location --insecure http://www.mr511.de/software/libelf-0.8.13.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   232  100   232    0     0    260      0 --:--:-- --:--:-- --:--:--   259
  0     0    0     0    0     0      0      0 --:--:--  0:00:03 --:--:--     0
curl: (52) Empty reply from server
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.lede-project.org/libelf-0.8.13.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  145k  100  145k    0     0  53562      0  0:00:02  0:00:02 --:--:-- 53543
. /home/creedowl/lede/include/shell.sh; gzip -dc /home/creedowl/lede/dl/libelf-0.8.13.tar.gz | tar -C /home/creedowl/lede/build_dir/host/libelf-0.8.13/.. -xf - 
[ ! -d ./src/ ] || cp -fpR ./src/* /home/creedowl/lede/build_dir/host/libelf-0.8.13
touch /home/creedowl/lede/build_dir/host/libelf-0.8.13/.preparedcc35b317a587bf834a38d4cf62b2ba50_6664517399ebbbc92a37c5bb081b5c53
(cd /home/creedowl/lede/build_dir/host/libelf-0.8.13/; bash ./configure --target=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/home/creedowl/lede/staging_dir/host --exec-prefix=/home/creedowl/lede/staging_dir/host --sysconfdir=/home/creedowl/lede/staging_dir/host/etc --localstatedir=/home/creedowl/lede/staging_dir/host/var --sbindir=/home/creedowl/lede/staging_dir/host/bin --disable-shared --enable-elf64; )
creating cache ./config.cache
bash: line 1: 61595 Segmentation fault      (core dumped) bash ./configure --target=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/home/creedowl/lede/staging_dir/host --exec-prefix=/home/creedowl/lede/staging_dir/host --sysconfdir=/home/creedowl/lede/staging_dir/host/etc --localstatedir=/home/creedowl/lede/staging_dir/host/var --sbindir=/home/creedowl/lede/staging_dir/host/bin --disable-shared --enable-elf64
make[3]: *** [Makefile:52: /home/creedowl/lede/build_dir/host/libelf-0.8.13/.configured] Error 139
make[3]: Leaving directory '/home/creedowl/lede/tools/libelf'
time: tools/libelf/compile#0.17#0.30#6.52
make[2]: *** [tools/Makefile:157: tools/libelf/compile] Error 2
make[2]: Leaving directory '/home/creedowl/lede'
make[1]: *** [tools/Makefile:153: /home/creedowl/lede/staging_dir/target-x86_64_musl/stamp/.tools_compile_yynyyyyynyyyyynyynnnyyyynyyyyyyyyyyyyyyynyyynynnyyynnyy] Error 2
make[1]: Leaving directory '/home/creedowl/lede'
make: *** [/home/creedowl/lede/include/toplevel.mk:218：world] 错误 2
```

lyin888, 2019-03-21:
[libelf-0.8.13.tar.gz](https://github.com/coolsnowwolf/lede/files/2991694/libelf-0.8.13.tar.gz)

lyin888, 2019-03-21:
下载不下来的文件去google搜索下载下来放在dl目录，然后重新编译.

Creedowl, 2019-03-21:
> 下载不下来的文件去google搜索下载下来放在dl目录，然后重新编译.

解决了，谢谢

closed 2019-03-22 by coolsnowwolf